moveit/moveit_tutorials #301: Fail on the second step at Move Group Python Interface tutorial

stanislauhalauko, 2019-03-24:
### Description

Move Group Python Interface fail. Can't go ahead in this tutorial after the first step.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04

### Steps to reproduce
I tried to follow [Move Group Python Interface tutorial ](https://ros-planning.github.io/moveit_tutorials/doc/move_group_python_interface/move_group_python_interface_tutorial.html). I ran _demo.launch_ and next  _move_group_python_interface_tutorial.py_ as it is written in tutorial. After first Enter I see expected result, but at the second step I see:

`Traceback (most recent call last):
  File "/home/stanislaw/ROS/ws_moveit/src/moveit_tutorials/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py", line 510, in <module>
    main()
  File "/home/stanislaw/ROS/ws_moveit/src/moveit_tutorials/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py", line 464, in main
    tutorial.go_to_joint_state()
  File "/home/stanislaw/ROS/ws_moveit/src/moveit_tutorials/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py", line 174, in go_to_joint_state
    move_group.go(joint_goal, wait=True)
  File "/home/stanislaw/ROS/ws_moveit/src/moveit/moveit_commander/src/moveit_commander/move_group.py", line 498, in go
    self.set_joint_value_target(self.get_remembered_joint_values()[joints])
TypeError: unhashable type: 'list'
`
and process fail.

Worth to notice:
1. When I run the _rosrun ..._  command, appears 
`Failed to import pyassimp, see https://github.com/ros-planning/moveit/issues/86 for more info
`
I have checked that issue and after checked my pyassimp version and it is 4.1.4 - theoretically shoud work properly.
I tried to import pyassimp in pure python2 and it faild, but when I imported it to python3 on my machine it was working.
I tried to run _move_group_python_interface_tutorial.py_ with command _python ....py_ and _python3 ....py_. In both cases it faild.

2. In the first step of .py script I get a warning:
```
[ WARN] [1553437979.464594324]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/panda_arm/kinematics_solver_attempts' from your configuration.
```

Where could be a problem?

shyamashi, 2019-05-07:
same error here. Any updates please. 
This was working properly for me until I pulled the latest version of moveit. Is going back to a previous version is the only solution?

rhaschke, 2019-05-07:
@stanislauhalauko I'm confused. Can you check your MoveIt version once again?
The warning "kinematics solver doesn't support #attempts" is due to #1288, which was only merged into the melodic/master branch, but not into kinetic.

shyamashi, 2019-05-07:
In catkin_ws\src\moveit ```git describe --tags``` gives 1.0.1-28-g7793148.
I followed https://moveit.ros.org/install/source/ to build moveit from source. ```go_to_joint_state()``` doesnt work and results in the error  ```TypeError: unhashable type: 'list' ```

stanislauhalauko, 2019-05-07:
@rhaschke I have few packages connected to MoveIt. Version of which one I should check?
[image]

E.g. version of moveit_core is 0.9.15

rhaschke, 2019-05-07:
This is actually the master branch. On Bionic I wasn't able to reproduce your issue using the master branch.
I will try to reproduce it on Xenial tomorrow, but I cannot promise I will find time for it...

rhaschke, 2019-05-07:
How did you figure the version of e.g. moveit_core? The source folder is the master branch, i.e. > 1.0.1.

rhaschke, 2019-05-08:
I was able to reproduce on Bionic and will prepare a fix.

rhaschke, 2019-05-08:
Issue fixed in MoveIt (https://github.com/ros-planning/moveit/pull/1459).

closed 2019-05-08 by rhaschke

stanislauhalauko, 2019-05-08:
@rhaschke Thank you a lot! Now it works. 
Btw. I've checked version of _moveit_core_ by the command   _rosversion moveit_core_