moveit/moveit_tutorials #303: Using moveit for 2 robots -- Didn't received robot state (joint angles)

XiaoshanLin9701, 2019-03-29:
### Description

I am using moveit for motion planning for 2 robots under different namespace: robot1 and robot2, but it seems moveit python interface only listen to **/joint_states** instead of **/robot1/joint_states** and **/robot2/joint_states**. So I have problem moving the robots to specfic joint states

### Your environment
* ROS Distro: Kinetic
* OS Version:  Ubuntu 16.04

### Steps to reproduce
I have a robot.urdf and spawn two robots under different namespace(/robot1 /robot2) in launch file. And then I launch the demo.launch file for the two robots:
```
<launch>
    <group ns="robot1">
        <include file="$(find kuka_moveit)/launch/demo.launch">
        </include>
    </group>
    <group ns="robot2">
        <include file="$(find kuka_moveit)/launch/demo.launch">
        </include>
    </group>
</launch>
```
And then I run move_group_python_interface_tutorial.py and make some modification.
I modify the following codes
```
robot = moveit_commander.RobotCommander()
group_name = "panda_arm"
group = moveit_commander.MoveGroupCommander(group_name)
```
to
```
robot1 = moveit_commander.RobotCommander("/robot1/robot_description")
robot2 = moveit_commander.RobotCommander("/robot2/robot_description")
group_name = "kuka_arm"
group1 = moveit_commander.MoveGroupCommander(group_name, "/robot1/robot_description"，“robot1”)
group2 = moveit_commander.MoveGroupCommander(group_name, "/robot2/robot_description", "robot2")
```
But when I rosrun  move_group_python_interface_tutorial.py, the function group1.group.get_current_joint_values() throw an error: Didn't received robot state (joint angles) with recent timestamp within 1 seconds.
Check clock synchronization if your are running ROS across multiple machines!
### Expected behaviour
It seems that moveit group is listening to /joint_states, but I have **/robot1/joint_states** and **/robot2/joint_states**, how to enable group1 subscribe to /robot1/joint_states, and make sure group2  subscribe to  /robot2/joint_states  

Your reply is much appreciated!

rhaschke, 2019-03-29:
`MoveGroupCommander` [has a namespace argument](https://github.com/ros-planning/moveit/blob/ff50476c4070eb86d0a70aa39281d5805db13fa5/moveit_commander/src/moveit_commander/move_group.py#L51), which you need to specify:
`moveit_commander.RobotCommander(ns="robot1")`

closed 2019-03-29 by rhaschke

XiaoshanLin9701, 2019-03-30:
Problem fixed! Thank you very much

Murtaza786-oss, 2019-09-18:
I am facing a same issue with controlling two robots. Please help me out here,

This is my launch which includes three python files vision (ur5_vision.py), motion planning (ur5_mp.py) and gripper (ur5_gripper.py) 
The motion planning file is the one which uses moveit interface to move the robot
With just one robot everything works fine but when I try with two robots I get the following error

Two Robots gets launched successfully but they don't perform anythin

I have attached my codes here in the following sequence
1) The original launch file (single UR5)
2) New Launch File (Two UR5)
3) Motion Planning (UR5_mp.py)

```
<?xml version="1.0"?>
<launch>
  <param name="red_box_path" type="str" value="$(find ur5_notebook)/urdf/red_box.urdf"/>

  <arg name="limited" default="true"/>
  <arg name="paused" default="false"/>
  <arg name="gui" default="true"/>
  <arg name="debug" default="false" />
  <arg name="sim" default="true" />

  <!-- startup simulated world -->
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" default="worlds/empty.world"/>
    <arg name="paused" value="$(arg paused)"/>
    <arg name="gui" value="$(arg gui)"/>
  </include>

  <!-- spwan ur5 -->
  <!-- send robot urdf to param server -->
  <include file="$(find ur_description)/launch/ur5_upload.launch">
    <arg name="limited" value="$(arg limited)"/>
  </include>

  <!-- push robot_description to factory and spawn robot in gazebo -->
  <node name="spawn_gazebo_model" pkg="gazebo_ros" type="spawn_model" args="-urdf -param robot_description -model robot -z 0.2 -y 0.7" respawn="false" output="screen" />

  <include file="$(find ur_gazebo)/launch/controller_utils.launch"/>

  <rosparam file="$(find ur_gazebo)/controller/arm_controller_ur5.yaml" command="load"/>
  <node name="arm_controller_spawner" pkg="controller_manager" type="controller_manager" args="spawn arm_controller" respawn="false" output="screen"/>

  <!-- Remap follow_joint_trajectory -->
  <remap if="$(arg sim)" from="follow_joint_trajectory" to="arm_controller/follow_joint_trajectory"/>

  <!-- Launch moveit -->
  <include file="$(find ur5_moveit_config)/launch/move_group.launch">
    <arg name="limited" default="$(arg limited)"/>
    <arg name="debug" default="$(arg debug)" />
  </include>

  <!-- spawn the conveyor_belt in gazebo -->
  <node name="spawn_conveyor_belt" pkg="gazebo_ros" type="spawn_model" args="-file $(find ur5_notebook)/urdf/conveyor_belt.urdf -urdf -model conveyor_belt" />
  <!-- spawn the conveyor_belt in gazebo -->
  <node name="bin" pkg="gazebo_ros" type="spawn_model" args="-file $(find ur5_notebook)/urdf/bin.urdf -urdf -model bin -y 0.8 -x -0.5 -z 0.05" />
  <!-- the red blocks spawner node -->
  <node name="blocks_spawner" pkg="ur5_notebook" type="blocks_spawner" output="screen" />

  <!-- the cylinder poses publisher node -->
  <node name="blocks_poses_publisher" pkg="ur5_notebook" type="blocks_poses_publisher" output="screen" />

  <!-- the cylinder poses publisher node -->
  <node name="ur5_vision" pkg="ur5_notebook" type="ur5_vision.py" output="screen" />
  <node name="ur5_mp" pkg="ur5_notebook" type="ur5_mp.py" output="screen" />
  <node name="ur5_gripper" pkg="ur5_notebook" type="ur5_gripper.py" output="screen"/>
  <!-- for ros control to be used with scara robot -->
  <!--   <param name="/scara_robot_left/robot_description" textfile="$(find two_scara_collaboration)/urdf/scara_robot_left.urdf" />-->
[... 455 lines cut ...]
               The jump_threshold specifies the maximum distance in configuration space between consecutive points
               in the resultingpath. The return value is a tuple: a fraction of how much of the path was followed,
               the actual RobotTrajectory.

            """
            plan, fraction = self.arm.compute_cartesian_path(self.waypoints, 0.01, 0.0, True)

            # plan = self.arm.plan()

            # If we have a complete plan, execute the trajectory
            if 1-fraction < 0.2:
                rospy.loginfo("Path computed successfully. Moving the arm.")
                num_pts = len(plan.joint_trajectory.points)
                rospy.loginfo("\n# intermediate waypoints = "+str(num_pts))
                self.arm.execute(plan)
                rospy.loginfo("Path execution complete.")
            else:
                rospy.loginfo("Path planning failed")
        # print self.points

mp=ur5_mp()

rospy.spin()

```