moveit/moveit_tutorials #307: Internal Compiler Error on Clean System install

jbcpollak, 2019-04-01:
### Description

I have a brand new Ubuntu 18.04 system with Ros Melodic installed. Trying to follow the MoveIt! tutorial here (https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html) I get a GCC internal compiler error when running `catkin build`

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binaries
* If binary, which release version?

Looks like most moveit binary packages have the version `1.0.1-0bionic.20190320.191310 amd64`
* If source, which git commit or tag?
The tutorial is at:
git https://github.com/ros-planning/moveit_tutorials.git#adccd9ebf5f06b1de207310000501c54881b97b8

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour

Should build cleanly with `catkin build`

### Backtrace or Console output

```
Scanning dependencies of target robot_model_and_robot_state_tutorial
Scanning dependencies of target ros_api_tutorial
[  3%] Building CXX object doc/kinematics/CMakeFiles/ros_api_tutorial.dir/src/ros_api_tutorial.cpp.o
[  7%] Building CXX object doc/robot_model_and_robot_state/CMakeFiles/robot_model_and_robot_state_tutorial.dir/src/robot_model_and_robot_state_tutorial.cpp.o
[ 10%] Linking CXX executable /home/parallels/src/moveit/devel/.private/moveit_tutorials/lib/moveit_tutorials/robot_model_and_robot_state_tutorial
[ 10%] Built target robot_model_and_robot_state_tutorial
Scanning dependencies of target planning_scene_tutorial
[ 14%] Building CXX object doc/planning_scene/CMakeFiles/planning_scene_tutorial.dir/src/planning_scene_tutorial.cpp.o
[ 17%] Linking CXX executable /home/parallels/src/moveit/devel/.private/moveit_tutorials/lib/moveit_tutorials/ros_api_tutorial
[ 17%] Built target ros_api_tutorial
Scanning dependencies of target planning_scene_ros_api_tutorial
[ 21%] Building CXX object doc/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/src/planning_scene_ros_api_tutorial.cpp.o
[ 25%] Linking CXX executable /home/parallels/src/moveit/devel/.private/moveit_tutorials/lib/moveit_tutorials/planning_scene_tutorial
[ 25%] Built target planning_scene_tutorial
Scanning dependencies of target motion_planning_api_tutorial
[ 28%] Building CXX object doc/motion_planning_api/CMakeFiles/motion_planning_api_tutorial.dir/src/motion_planning_api_tutorial.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
doc/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/build.make:62: recipe for target 'doc/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/src/planning_scene_ros_api_tutorial.cpp.o' failed
make[2]: *** [doc/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/src/planning_scene_ros_api_tutorial.cpp.o] Error 4
CMakeFiles/Makefile2:4325: recipe for target 'doc/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/all' failed
make[1]: *** [doc/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 32%] Linking CXX executable /home/parallels/src/moveit/devel/.private/moveit_tutorials/lib/moveit_tutorials/motion_planning_api_tutorial
[ 32%] Built target motion_planning_api_tutorial
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

jbcpollak, 2019-04-01:
I tried building moveit from source as well, and got the same error on a different package:

output from `catkin build`:

```
...............................................................................
Finished  <<< moveit_simple_controller_manager                 [ 28.0 seconds ]
Starting  >>> moveit_chomp_optimizer_adapter                                   
_______________________________________________________________________________
Warnings   << moveit_chomp_optimizer_adapter:symlink /home/parallels/src/moveit/logs/moveit_chomp_optimizer_adapter/build.symlink.000.log
Warning: Cannot symlink from /home/parallels/src/moveit/devel/.private/moveit_chomp_optimizer_adapter/local_setup.sh to existing file /home/parallels/src/moveit/devel/./local_setup.sh
Warning: Source hash: 7e8e77bcf3aef589bf3accfd5e5b37b4
Warning: Dest hash: 9cb63592d776bb163f426a8bb307ca01
...............................................................................
Finished  <<< moveit_chomp_optimizer_adapter                   [ 17.4 seconds ]
Starting  >>> moveit_ros_planning                                              
_______________________________________________________________________________
Errors     << moveit_ros_planning:make /home/parallels/src/moveit/logs/moveit_ros_planning/build.make.000.log
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [planning_scene_monitor/CMakeFiles/demo_scene.dir/demos/demo_scene.cpp.o] Error 4
make[1]: *** [planning_scene_monitor/CMakeFiles/demo_scene.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/parallels/src/moveit/build/moveit_ros_planning; catkin build --get-env moveit_ros_planning | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << moveit_ros_planning:make                         [ Exited with code 2 ]
Failed    <<< moveit_ros_planning                              [ 7 minutes and 7.3 seconds ]
```

and the log file:

```
Scanning dependencies of target moveit_kinematics_speed_and_validity_evaluator
[ 56%] Building CXX object planning_components_tools/CMakeFiles/moveit_kinematics_speed_and_validity_evaluator.dir/src/kinematics_speed_and_validity_evaluator.cpp.o
[ 58%] Linking CXX executable /home/parallels/src/moveit/devel/.private/moveit_ros_planning/lib/moveit_ros_planning/moveit_kinematics_speed_and_validity_evaluator
[ 58%] Built target moveit_kinematics_speed_and_validity_evaluator
[ 60%] Building CXX object planning_scene_monitor/CMakeFiles/moveit_planning_scene_monitor.dir/src/current_state_monitor.cpp.o
[ 62%] Building CXX object planning_scene_monitor/CMakeFiles/moveit_planning_scene_monitor.dir/src/trajectory_monitor.cpp.o
[ 63%] Linking CXX shared library /home/parallels/src/moveit/devel/.private/moveit_ros_planning/lib/libmoveit_planning_scene_monitor.so
[ 63%] Built target moveit_planning_scene_monitor
Scanning dependencies of target demo_scene
Scanning dependencies of target moveit_print_planning_scene_info
[ 67%] Building CXX object planning_scene_monitor/CMakeFiles/demo_scene.dir/demos/demo_scene.cpp.o
[ 67%] Building CXX object planning_components_tools/CMakeFiles/moveit_print_planning_scene_info.dir/src/print_planning_scene_info.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [planning_scene_monitor/CMakeFiles/demo_scene.dir/demos/demo_scene.cpp.o] Error 4
planning_scene_monitor/CMakeFiles/demo_scene.dir/build.make:62: recipe for target 'planning_scene_monitor/CMakeFiles/demo_scene.dir/demos/demo_scene.cpp.o' failed
CMakeFiles/Makefile2:3362: recipe for target 'planning_scene_monitor/CMakeFiles/demo_scene.dir/all' failed
make[1]: *** [planning_scene_monitor/CMakeFiles/demo_scene.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

rhaschke, 2019-04-01:
Are you running out-of-memory during building? This is a typical case making gcc fail.
catkin attempts to build on several cores in parallel, which can easily exhaust your memory. If this is the case, please try to limit the number of parallel jobs via `-j 1`.

jbcpollak, 2019-04-01:
@rhaschke - That was it! I was compiling in a VM with 1gb of ram. Bumping it to 4gb fixed it. Thanks!

closed 2019-04-01 by jbcpollak