moveit/moveit_tutorials #312: motion_planning_api Tutorial Hangs "Waiting for complete state from topic "

rmerriam, 2019-04-07:
### Description

The motion_planning_api Tutorial Hangs "Waiting for complete state from topic "

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04
* Binary
* Latest updated

### Steps to reproduce
Terminal 1: roslaunch panda_moveit_config demo.launch
Terminal 2: roslaunch moveit_tutorials motion_planning_api_tutorial.launch

### Expected behaviour
Tell us what should happen

### Backtrace or Console output

[exec_motion_planning_api.txt](https://github.com/ros-planning/moveit_tutorials/files/3051123/exec_motion_planning_api.txt)
[rviz_motion_planning_api.txt](https://github.com/ros-planning/moveit_tutorials/files/3051124/rviz_motion_planning_api.txt)

Dale-Koenig, 2019-05-14:
I fixed this by changing the virtual joint type in panda_moveit_config/config/panda_arm.xacro from "floating" to "fixed"

rhaschke, 2019-05-14:
Assuming a fixed joint is one option to handle this.
The better way would be to _publish_ the corresponding transform, e.g. with a `static transform publisher`.

rhaschke, 2019-05-14:
@mlautman, actually the roslaunch files should run the static transform publisher, shouldn't they?

BryceStevenWilley, 2019-05-14:
This should have been fixed in https://github.com/ros-planning/panda_moveit_config/pull/23, but I have seen a lot of issues popping up about it lately. @Dale-Koenig, could you confirm that getting the latest commit from `panda_moveit_config` fixes this issue, without having to change the floating virtual joint type to fixed?

Dale-Koenig, 2019-05-15:
@BryceStevenWilley The `virtual_joint_broadcaster_1` node exists but the same problem occurs.

rhaschke, 2019-08-30:
Fixed via https://github.com/ros-planning/moveit_tutorials/pull/387.

closed 2019-08-30 by rhaschke