moveit/moveit_tutorials #314: Moveit_tutorials can't be catkin built because of Eigen3 with the version of kinetic

mengzhilinxi2018, 2019-04-17:
### Description

I met a problrm that Movtit_tutorials can't be catkin built because of Eigen3. My ROS system is Kinetic, and the Eigen3 version is 3.3~beta 1-2. From the error log file, I think the version of Moveit_tutorials is not compatible with the version of Eigen3. Because the error log file shows the problem is from  Eigen::transform() function. Furthermore, I check the tranform.h file from Eigen3 lib, the Eigen::Transform() function is forbidden to use, because this function may cause error. But the error log file show that the movtit_tutorial have call the Eigen::transform() function, so it make a mistake.

### Your environment
* ROS Distro: [Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source

* If source, which git commit or tag?
https://github.com/ros-planning/moveit_tutorials.git

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.
First, follow the web link http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/getting_started/getting_started.html#getting-started
 to install moveit and moveit_tutorials. And in the past I have use the command "sudo apt-get install libeigen3-dev" to install Eigen3 lib. Next, we you catkin build in the ws_moveit, there will be error log shown in the following. 
### Expected behaviour
Tell us what should happen
The error log show in the following.
_______________________________________________________________________________
Errors     << moveit_tutorials:make /home/menghui/ws_moveit/logs/moveit_tutorials/build.make.010.log
In file included from /usr/include/eigen3/Eigen/Core:142:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model_group.h:41,
                 from /opt/ros/kinetic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model_loader/robot_model_loader.h:41,
                 from /home/menghui/ws_moveit/src/moveit_tutorials/doc/robot_model_and_robot_state/src/robot_model_and_robot_state_tutorial.cpp:40:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/menghui/ws_moveit/src/moveit_tutorials/doc/robot_model_and_robot_state/src/robot_model_and_robot_state_tutorial.cpp:118:102:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:33:40: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:334:5: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
In file included from /usr/include/eigen3/Eigen/Core:142:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model_group.h:41,
                 from /opt/ros/kinetic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model_loader/robot_model_loader.h:41,
                 from /home/menghui/ws_moveit/src/moveit_tutorials/doc/interactivity/include/interactivity/interactive_robot.h:43,
                 from /home/menghui/ws_moveit/src/moveit_tutorials/doc/interactivity/src/interactive_robot.cpp:39:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/menghui/ws_moveit/src/moveit_tutorials/doc/interactivity/src/interactive_robot.cpp:93:32:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:33:40: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:334:5: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
In file included from /usr/include/eigen3/Eigen/Core:142:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model_group.h:41,
                 from /opt/ros/kinetic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model_loader/robot_model_loader.h:41,
                 from /home/menghui/ws_moveit/src/moveit_tutorials/doc/state_display/src/state_display_tutorial.cpp:40:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/menghui/ws_moveit/src/moveit_tutorials/doc/state_display/src/state_display_tutorial.cpp:101:114:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:33:40: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:334:5: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
make[2]: *** [doc/robot_model_and_robot_state/CMakeFiles/robot_model_and_robot_state_tutorial.dir/src/robot_model_and_robot_state_tutorial.cpp.o] Error 1
make[1]: *** [doc/robot_model_and_robot_state/CMakeFiles/robot_model_and_robot_state_tutorial.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
[... 3 lines cut ...]
make[1]: *** [doc/interactivity/CMakeFiles/interactivity_utils.dir/all] Error 2
/home/menghui/ws_moveit/src/moveit_tutorials/doc/move_group_interface/src/move_group_interface_tutorial.cpp: In function ‘int main(int, char**)’:
/home/menghui/ws_moveit/src/moveit_tutorials/doc/move_group_interface/src/move_group_interface_tutorial.cpp:109:24: error: ‘class moveit::planning_interface::MoveGroupInterface’ has no member named ‘getJointModelGroupNames’
   std::copy(move_group.getJointModelGroupNames().begin(), move_group.getJointModelGroupNames().end(),
                        ^
/home/menghui/ws_moveit/src/moveit_tutorials/doc/move_group_interface/src/move_group_interface_tutorial.cpp:109:70: error: ‘class moveit::planning_interface::MoveGroupInterface’ has no member named ‘getJointModelGroupNames’
   std::copy(move_group.getJointModelGroupNames().begin(), move_group.getJointModelGroupNames().end(),
                                                                      ^
make[2]: *** [doc/move_group_interface/CMakeFiles/move_group_interface_tutorial.dir/src/move_group_interface_tutorial.cpp.o] Error 1
make[1]: *** [doc/move_group_interface/CMakeFiles/move_group_interface_tutorial.dir/all] Error 2
make: *** [all] Error 2
cd /home/menghui/ws_moveit/build/moveit_tutorials; catkin build --get-env moveit_tutorials | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << moveit_tutorials:make                            [ Exited with code 2 ]
Failed    <<< moveit_tutorials                                 [ 5.7 seconds ] 
[build] Summary: 29 of 30 packages succeeded.                                  
[build]   Ignored:   5 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 10.6 seconds total.         

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

gavanderhoorn, 2019-04-17:
Cross-post of [MoveIt_tutorials can't be catkin built by Eigen3](https://answers.ros.org/question/321250) on ROS Answers.

mlautman, 2019-04-22:
As best as I can tell, you are building the master branch of the tutorials when you should be building the kinetic-devel branch. 

> And in the past I have use the command "sudo apt-get install libeigen3-dev" to install Eigen3 lib

You should not be doing this manually. the `rosdep install` command should take care of installing all dependencies for you.

mlautman, 2019-04-22:
Re-open if you continue to have issues

closed 2019-04-22 by mlautman

mengzhilinxi2018, 2019-04-23:
Okay. Thank you for your answer. I will follow it.