moveit/moveit_tutorials #318: Can't attach object 'box' to link 'panda_link8' in Move Group Python Interface

gtatiya, 2019-04-29:
### Description
I am executing `rosrun moveit_tutorials move_group_python_interface_tutorial.py` and on the 6th step when it is supposed to add a box at the location of the Panda end effector, noting happens. And for the 7th step, I get error: **Attempting to attach object 'box' to link 'panda_link8' but no geometry specified and such an object does not exist in the collision world**. Then I get error: **Attached body 'box' not found**. Basically, it does everything in this video https://youtu.be/3MA5ebXPLsc , but without the block. **How to add the block?**

Overview of your issue here.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source or Binary build?
https://github.com/ros-planning/moveit_tutorials/tree/kinetic-devel/

### Steps to reproduce
`roslaunch panda_moveit_config demo.launch`
`rosrun moveit_tutorials move_group_python_interface_tutorial.py`

### Backtrace or Console output
You can start planning now!

[ INFO] [1556561818.157618836]: Stereo is NOT SUPPORTED
[ INFO] [1556561818.157718220]: OpenGl version: 4.5 (GLSL 4.5).
[ INFO] [1556561818.362097277]: Loading robot model 'panda'...
[ INFO] [1556561818.784820296]: Loading robot model 'panda'...
[ INFO] [1556561818.846164583]: Loading robot model 'panda'...
[ INFO] [1556561818.905782985]: Starting scene monitor
[ INFO] [1556561818.909110725]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1556561841.368812097]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1556561841.369020601]: Planning attempt 1 of at most 1
[ INFO] [1556561841.370242689]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1556561841.370871923]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1556561841.381802425]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1556561841.381834572]: Solution found in 0.011215 seconds
[ INFO] [1556561841.383212112]: SimpleSetup: Path simplification took 0.001305 seconds and changed from 3 to 2 states
[ INFO] [1556561841.470457585]: Fake execution of trajectory
[ WARN] [1556561841.568710859]: The complete state of the robot is not yet known.  Missing virtual_joint
[ WARN] [1556561842.668632604]: The complete state of the robot is not yet known.  Missing virtual_joint
[ INFO] [1556561842.868690130]: Completed trajectory execution with status SUCCEEDED ...
[ INFO] [1556561842.985073477]: Received event 'stop'
[ INFO] [1556561844.268695763]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1556561844.268962443]: Planning attempt 1 of at most 1
[ INFO] [1556561844.269469836]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1556561844.269625883]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1556561844.290463445]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1556561844.290508211]: Solution found in 0.020939 seconds
[ INFO] [1556561844.293952456]: SimpleSetup: Path simplification took 0.003413 seconds and changed from 3 to 2 states
[ INFO] [1556561844.368853058]: Fake execution of trajectory
[ WARN] [1556561844.468649355]: The complete state of the robot is not yet known.  Missing virtual_joint
[ WARN] [1556561845.469017638]: The complete state of the robot is not yet known.  Missing virtual_joint
[ INFO] [1556561846.468616334]: Completed trajectory execution with status SUCCEEDED ...
[ INFO] [1556561846.495568756]: Received event 'stop'
[ INFO] [1556561849.270222016]: Received request to compute Cartesian path
[ INFO] [1556561849.270373329]: Attempting to follow 3 waypoints for link 'panda_link8' using a step of 0.010000 m and jump threshold 0.000000 (in global reference frame)
[ INFO] [1556561849.277641065]: Computed Cartesian path with 46 points (followed 100.000000% of requested trajectory)
[ INFO] [1556561853.047299863]: Execution request received
[ INFO] [1556561853.068808178]: Fake execution of trajectory
[ WARN] [1556561853.168581106]: The complete state of the robot is not yet known.  Missing virtual_joint
[ WARN] [1556561854.168627449]: The complete state of the robot is not yet known.  Missing virtual_joint
[ WARN] [1556561855.283499501]: The complete state of the robot is not yet known.  Missing virtual_joint
[ INFO] [1556561855.468660088]: Completed trajectory execution with status SUCCEEDED ...
[ INFO] [1556561855.468758580]: Execution completed: SUCCEEDED
[ERROR] [1556561863.246691158]: Attempting to attach object 'box' to link 'panda_link8' but no geometry specified and such an object does not exist in the collision world
[ INFO] [1556561867.971751022]: Received request to compute Cartesian path
[ INFO] [1556561867.971912221]: Attempting to follow 3 waypoints for link 'panda_link8' using a step of 0.010000 m and jump threshold 0.000000 (in global reference frame)
[ INFO] [1556561867.977062609]: Computed Cartesian path with 45 points (followed 100.000000% of requested trajectory)
[ INFO] [1556561867.980222843]: Execution request received
[ INFO] [1556561868.068844837]: Fake execution of trajectory
[ WARN] [1556561868.168641663]: The complete state of the robot is not yet known.  Missing virtual_joint
[ WARN] [1556561869.268681279]: The complete state of the robot is not yet known.  Missing virtual_joint
[ INFO] [1556561870.368583455]: Completed trajectory execution with status SUCCEEDED ...
[ INFO] [1556561870.368670544]: Execution completed: SUCCEEDED
[ERROR] [1556561871.630572004]: Attached body 'box' not found
[ WARN] [1556561875.731115697]: MessageFilter [target=/world ]: Discarding message from [/move_group_python_interface_tutorial_2748_1556561836901] due to empty frame_id.  This message will only print once.

mlautman, 2019-04-29:
works for me.

 [image]

You should check the output of this line: https://github.com/ros-planning/moveit_tutorials/blob/65af8798d8ad9ac0cc10416f451136b56d910115/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py#L466

If it returns false, then the box isn't being added to the planning scene.

I would recommend installing the latest versions of panda_moveit_config, moveit, and moveit_tutorials

closed 2019-04-29 by mlautman

gtatiya, 2019-04-29:
@mlautman Output of `tutorial.add_box()` is `False`. I installed the latest versions of panda_moveit_config, moveit, and moveit_tutorials today itself. So, I'm using the latest versions. Could you please help me debug this problem.

BryceStevenWilley, 2019-04-30:
Hey @gtatiya, you can find some more direct help if you post this on https://answers.ros.org.

gavanderhoorn, 2019-05-01:
ROS Answers post: [Can't attach object 'box' to link 'panda_link8' in Move Group Python Interface](https://answers.ros.org/question/322220).

gtatiya, 2019-05-01:
@mlautman, @BryceStevenWilley, @gavanderhoorn I posted this question on https://answers.ros.org/question/322220/cant-attach-object-box-to-link-panda_link8-in-move-group-python-interface/ but no one is able to help me. A friend also tried this on this computer, but he is also facing same problem. This is important for my project. Could you please help me out.

gtatiya, 2019-05-03:
@mlautman, @BryceStevenWilley, @gavanderhoorn In line 366 (https://github.com/ros-planning/moveit_tutorials/blob/65af8798d8ad9ac0cc10416f451136b56d910115/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py#L366) `box_name` is assigned to `self.box_name`. But, why `scene` is not assigned to `self.scene`? I have tried `self.scene = scene` , but it still does not work. Any idea what I should do?

martinakos, 2019-06-07:
I have the same problem. I wonder if it has to do with pyassimp. I got rid off a pyassimp import error following [this](https://answers.ros.org/question/316515/cannot-add-mesh-files-to-moveit/). But I still have the problem not being able to attach the box. tutorial.add_box() is False for me too.