moveit/moveit_tutorials #321: Trouble executing "rosrun moveit_tutorials move_group_python_interface_tutorial.py"

IndustrialEngStudent, 2019-05-06:
### Description
I am trying to follow the Move Group Python Interface tutorial, I follow the exact steps from the "Getting Started" tutorial (http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/getting_started/getting_started.html). When trying to run moveit_tutorials move_group_python_interface_tutorial.py node is says that the package is not found. I think it has something to do with the fact I now have two workspaces "~/catkin_ws & ~/ws_moveit. However, I do not know what to do to fix it. I have executed "catkin build" and in the output of this command it does show the package "moveit_tutorials":
```
laurah@laurah-HP-EliteBook-840-G1:~/ws_moveit$ catkin build
-----------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/kinetic
Workspace:                   /home/laurah/ws_moveit
-----------------------------------------------------------
Build Space:        [exists] /home/laurah/ws_moveit/build
Devel Space:        [exists] /home/laurah/ws_moveit/devel
Install Space:      [unused] /home/laurah/ws_moveit/install
Log Space:          [exists] /home/laurah/ws_moveit/logs
Source Space:       [exists] /home/laurah/ws_moveit/src
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------
[build] Found '2' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                           
Starting  >>> panda_moveit_config                                              
Finished  <<< panda_moveit_config                [ 0.1 seconds ]               
Starting  >>> moveit_tutorials                                                 
Finished  <<< moveit_tutorials                   [ 0.4 seconds ]               
[build] Summary: All 2 packages succeeded!                                     
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 0.6 seconds total.
```
Also I sourced it with "source devel/setup.bash". What can I do for the package to be found when using the rosrun command?

### Your environment
* ROS Distro: Kinetic
* OS Version: e.g. Ubuntu 16.04
* Source : git clone -b kinetic-devel https://github.com/ros-planning/moveit_tutorials.git

### Steps to reproduce
http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html

### Expected behaviour
rosrun  command should find package

### Backtrace or Console output
Input command (after catkin build & source):
laurah@laurah-HP-EliteBook-840-G1:~/ws_moveit$ rosrun moveit_tutorials move_group_python_interface_tutorial.py

Output:
[rospack] Error: package 'moveit_tutorials' not found

mlautman, 2019-05-29:
Thank you for opening this issue. Just to clarify, you are running `source devel/setup.bash` from within `/home/laurah/ws_moveit` correct? 

I have sometimes found that after building a package and sourcing the workspace, rospack still fails to find the new package. This seems to be the result of some caching in rospack. In my experience, opening a fresh terminal will sometimes fix the caching issue. Remove any `source devel/setup.bash` from your `.bashrc` and open a new terminal. Simply run `source /opt/ros/kinetic/setup.bash` then `source /home/laurah/ws_moveit/devel/setup.bash`. If you are still unable to run the `rosrun` command, then there is something very wrong with your setup.

Some other debug steps:
Are you able to run `roslaunch panda_moveit_config demo.launch`? if you run `roscd panda_moveit_config` does it take you to `/home/laurah/ws_moveit/src/panda_moveit_config` or `/opt/ros/kinetic/panda_moveit_config`? What about `roscd moveit_tutorials`?

Closing for now. Feel free to re-open if you still have this issue

closed 2019-05-29 by mlautman

YulinWang-001, 2020-03-03:
Hey, I have got similar questions, it keeps saying package 'moveit_tutorials' not found, and I have tried your method, and it didn't seem work, can you please help

jacob-02, 2021-07-03:
I am having a similar issue too. But if I try to run ` roscd panda-moveit_config` it says 

> roscd: No such package/stack 'panda-moveit_config'

Any help in this matter could be appreciated.

felixvd, 2021-07-03:
Surely that should be panda_moveit_config?

jacob-02, 2021-07-03:
Oh crap. I made some mistakes while copying. Thank you