moveit/moveit_tutorials #322: problem in package created by the moveit! setup assistant

cocodmdr, 2019-05-06:
### Description

demo.launch file is calling planning_context.launch which is looking for config package.
However config is a folder and not a package.

Also the urdf file that we use to create the robot arm is not put in folder config by the setup assistant. You have to put it manually. It could be done automatically

To resolve this first problem I changed in planning_context.launch line 9 from : 

`  <param if="$(arg load_robot_description)" name="$(arg robot_description)" textfile="$(find config)/arm.urdf"/>`

To : 

`  <param if="$(arg load_robot_description)" name="$(arg robot_description)" textfile="$(find 3d_printed_arm_config)/config/arm.urdf"/>`

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source or Binary build? binary build with sudo apt install ros-kinetic-moveit
* If binary, which release version? I don know

### Steps to reproduce
Create a new robot arm ( here called 3d_printed_arm_config) config files with the setup assistant and launch:
 roslaunch 3d_printed_arm_config demo.launch rviz_tutorial:=true

### Expected behaviour
This error appears : 
`WARNING: Package name "3d_printed_arm_config" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
... logging to /home/corentin/.ros/log/ff88820a-700d-11e9-8e99-5891cf9d3758/roslaunch-corentin-Dell-Precision-M3800-17479.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "3d_printed_arm_config" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 749, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 721, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 685, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 627, in _include_tag
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 665, in _recurse_load
    self._param_tag(tag, context, ros_config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 257, in _param_tag
    vals = self.opt_attrs(tag, context, ('value', 'textfile', 'binfile', 'command'))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 202, in opt_attrs
    return [self.resolve_args(tag_value(tag,a), context) for a in attrs]
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: config
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/corentin/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share`

BryceStevenWilley, 2019-05-07:
Tracking this back to the actual code gets you [here](https://github.com/ros-planning/moveit/blob/b3c4c6034e4d62d54199f22463f53417730a9d88/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/planning_context.launch#L9), which is set by [this line](https://github.com/ros-planning/moveit/blob/ff50476c4070eb86d0a70aa39281d5805db13fa5/moveit_setup_assistant/src/widgets/configuration_files_widget.cpp#L1061), which shows that the exact location should be given by the user. You summarized the user error yourself actually: if you leave out the package name in the URDF package path (just giving the relative path from the root dir of the package, like `config/robot.urdf` instead of `robot_description/config/robot.urdf`), the Setup Assistant will think that the `config` folder is the package name.

I don't think this is a bug, but want to be able to conclusively say so. @cocodmdr can you link the moveit_config you are using, or try creating the moveit config again, keeping the whole package path as suggested above?

cocodmdr, 2019-05-07:
So I am following this tutorial on Instructable : [https://www.instructables.com/id/ROS-MoveIt-Robotic-Arm/](url)
I figured out that the folder "config" where the urdf file is stored, contains a package.xml files and CMakeList.txt. This is why moveit setup assistant believes it is a package but in fact it is not recognized by ROS (a package inside a package does not work i suppose). 
I removed this package.xml and CMakeList.txt and used the moveit setup assistant again. This time it gives the good package path.
So I believe this issue is resolved. The problem comes from the folder provided in the tutorial.
Thank you for your fast response! I am sorry to have created this issue.

BryceStevenWilley, 2019-05-08:
After looking at the repo itself, you are correct, and the package is malformed. Thanks for your help.

closed 2019-05-08 by BryceStevenWilley