moveit/moveit_tutorials #324: Pick and place  issue

amranov, 2019-05-14:
Hi guys
i am trying to run the pick and place tutorial on real panda arm robot but i am getting this error 
[WARN] [1557823629.746422]: Controller Spawner error while taking down controllers: unable to connect to service: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
[WARN] [1557823629.751373]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 104] Connection reset by peer
Any aggression ?

mlautman, 2019-05-29:
@fwalch Any ideas here?

truhoang-ubtrobot, 2019-11-18:
@amranov, can you specify more details of your implementation and the robot you use?

closed 2019-12-24 by amranov