moveit/moveit_tutorials #329: can't load mjcf files (mujoco robot description) in moveit

kargarisaac, 2019-05-28:
### Description

Hi,
I want to use moveit with my robot described in mujoco format. I don't want to convert the mjcf file to urdf format. How can I use moveit for my work? I cannot load the mjcf file (with xml extension) in moveit setup assistant.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Binary build

### Steps to reproduce
load the xml file. for example
https://github.com/Feryal/a3c-mujoco/blob/master/jaco_arm/assets/jaco.xml 

### Expected behaviour
load the robot

BryceStevenWilley, 2019-05-28:
MoveIt extensively uses the URDF format, and has no plans to support Mujoco files. However, unless you specifically wrote your mjcf file yourself, I'm sure whatever robot you are using also has a provided URDF file. For example, for the linked jaco arm, there is Kinova_ros package (https://github.com/Kinovarobotics/kinova-ros/tree/master/kinova_description). 

In addition, from what I can tell, Mujoco itself [supports URDFs](http://mujoco.org/book/modeling.html).

closed 2019-05-28 by BryceStevenWilley

kargarisaac, 2019-05-28:
@BryceStevenWilley 
Yes, mujoco supports the urdf files. I have the urdf file of my robot too, but as in the mujoco book is mentioned, it is limited. 
http://mujoco.org/book/modeling.html#CURDF

rhaschke, 2019-05-28:
The link you are referring to, refers to using URDF files in MuJoCo. However, @BryceStevenWilley suggested to use .mjcf for MuJoCo and .urdf for MoveIt.

kargarisaac, 2019-05-28:
@rhaschke 
Yes and it says that urdf files with mujoco tags are limited. 
I don't know how much this limitation is. I just want to know is there any way to use moveit full capabilities with mujoco full capabilities. because if I want to use mujoco (mjcf), I have to implement controllers, trajectory planning and so on. I'm new in mujoco and moveit and any suggestion is welcome.

gavanderhoorn, 2019-05-28:
Looks like there is a ros integration for mujoco available here: [shadow-robot/mujoco_ros_pkgs](https://github.com/shadow-robot/mujoco_ros_pkgs). That was not too hard to find.

Note: I haven't used it, just asked google for "mujoco ros".

kargarisaac, 2019-05-28:
@gavanderhoorn 
yes I saw it, but I think it doesn't do what moveit does. I'm not sure

rhaschke, 2019-05-29:
@kargarisaac, I'm still not sure about your actual question: MuJoCo is a physics simulation engine and MoveIt is a planning framework. So both software packages handle different problems. In which fashion do you want to combine MoveIt and MuJoCo?
As proposed above, using .mjcf files for MuJoCo and .urdf files for MoveIt you can unleash the full power of both packages individually. The question is how and for which purpose do you plan to couple both?

kargarisaac, 2019-05-29:
@rhaschke, my model is in mjcf format and I want to load it in moveit setup assistant and I cannot. I want to control my robot (which is in mjcf format) and do some motion planning for it. control the end effector position and command it to grasp objects. Is it possible? 
Usually, I load urdf files in moveit setup assistant and do the configuration. Then I write a launch file and load the controller, moveit config, and any other node that I want. But for mjcf file, I don't know what to do. 
I think moveit needs the robot description, but I have the mjcf format and it cannot load it. is it possible to do planning using moveit and without using robot description?

gavanderhoorn, 2019-05-29:
> I want to load it in moveit setup assistant and I cannot. [..] Is it possible?

No. The MSA only supports loading `.urdf` and `.xacro`.

> I want to control my robot (which is in mjcf format) and do some motion planning for it. control the end effector position and command it to grasp objects.

This should be possible using [shadow-robot/mujoco_ros_pkgs](https://github.com/shadow-robot/mujoco_ros_pkgs) but you'll still need a robot description most likely (ie: `.urdf` or `.xacro`).

> I think moveit needs the robot description, 

yes.

> but I have the mjcf format and it cannot load it.

correct, as it's an unsupported format.

>  is it possible to do planning using moveit and without using robot description?

Technically? Yes.

pragmatically? Not really.

mlautman, 2019-05-29:
@kargarisaac A quick search over the MoveIt code base will show you that the URDF format is explicitly used by dynamics_solver, planning_scene, robot_model, and rdf_loader as well as throughout the moveit_setup_assistant. It would be feasible to modify these elements to use mjcf format although quite a bit of work. If you decide to make those modifications, please open a PR upstream so that the entire community can benefit from your work :)

gautams3, 2019-07-29:
> This should be possible using [shadow-robot/mujoco_ros_pkgs](https://github.com/shadow-robot/mujoco_ros_pkgs) but you'll still need a robot description most likely (ie: `.urdf` or `.xacro`).

`mujoco_ros_pkgs` also loads URDF not MJCF for robot description, so yes you'll need the `.urdf` or `.xacro`

https://github.com/shadow-robot/mujoco_ros_pkgs/blob/kinetic-devel/mujoco_ros_control/src/mujoco_ros_control.cpp#L65

wangcongrobot, 2019-10-17:
I think they have a try about the combine of Mujoco and MoveIt!.
https://github.com/gamleksi/mujoco_ros_control
https://www.youtube.com/playlist?list=PLwO8awctVMheqR0lSKW-R1jILyXx6po5N

zohar-feldman, 2019-10-23:
> `mujoco_ros_pkgs` also loads URDF not MJCF for robot description, so yes you'll need the `.urdf` or `.xacro` [...]

I have connected Mujoco with Moveit, and this is exactly the approach I took. I use urdf for moveit and an mjcf compiled from the urdf for the simulation