moveit/moveit_tutorials #339: Unable to 'catkin build' the example - Eigen3 issue

sguysc, 2019-06-17:
### Description

Following the tutorial [here](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html) and switching to "kinetic" wherever needed to suit my ROS version,
I recieved errors on the last step which is "catkin build".

Anyone knows how to fix the issue?

### Backtrace or Console output
This is what I got:

```
----------------------------------------------------------------------
Profile:                     default
Extendi
   ng:        [explicit] /opt/ros/kinetic
Workspace:                   /home/cc/Documents/ws_moveit
----------------------------------------------------------------------
Build Space:        [exists] /home/cc/Documents/ws_moveit/build
Devel Space:        [exists] /home/cc/Documents/ws_moveit/devel
Install Space:      [unused] /home/cc/Documents/ws_moveit/install
Log Space:         [missing] /home/cc/Documents/ws_moveit/logs
Source Space:       [exists] /home/cc/Documents/ws_moveit/src
DESTDIR:            [unused] None
----------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
----------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
----------------------------------------------------------------------
Workspace configuration appears valid.
----------------------------------------------------------------------
[build] Found '2' packages in 0.0 seconds.                                     
[build] Updating package table.                                                
Warning: generated devel space setup files have been deleted.
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                [ 1.4 seconds ]             
Starting  >>> panda_moveit_config                                              
Finished  <<< panda_moveit_config                  [ 1.4 seconds ]             
Starting  >>> moveit_tutorials                                                 
_______________________________________________________________________________
Errors     << moveit_tutorials:make /home/cc/Documents/ws_moveit/logs/moveit_tutorials/build.make.000.log
In file included from /usr/include/eigen3/Eigen/Core:297:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model_group.h:41,
                 from /opt/ros/kinetic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model_loader/robot_model_loader.h:41,
                 from /home/cc/Documents/ws_moveit/src/moveit_tutorials/doc/robot_model_and_robot_state/src/robot_model_and_robot_state_tutorial.cpp:40:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/cc/Documents/ws_moveit/src/moveit_tutorials/doc/robot_model_and_robot_state/src/robot_model_and_robot_state_tutorial.cpp:118:102:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:330:5: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
In file included from /usr/include/eigen3/Eigen/Core:297:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model_group.h:41,
                 from /opt/ros/kinetic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model_loader/robot_model_loader.h:41,
                 from /home/cc/Documents/ws_moveit/src/moveit_tutorials/doc/interactivity/include/interactivity/interactive_robot.h:43,
                 from /home/cc/Documents/ws_moveit/src/moveit_tutorials/doc/interactivity/src/interactive_robot.cpp:39:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/cc/Documents/ws_moveit/src/moveit_tutorials/doc/interactivity/src/interactive_robot.cpp:93:32:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:330:5: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
make[2]: *** [doc/robot_model_and_robot_state/CMakeFiles/robot_model_and_robot_state_tutorial.dir/src/robot_model_and_robot_state_tutorial.cpp.o] Error 1
make[1]: *** [doc/robot_model_and_robot_state/CMakeFiles/robot_model_and_robot_state_tutorial.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/cc/Documents/ws_moveit/src/moveit_tutorials/doc/move_group_interface/src/move_group_interface_tutorial.cpp: In function ‘int main(int, char**)’:
/home/cc/Documents/ws_moveit/src/moveit_tutorials/doc/move_group_interface/src/move_group_interface_tutorial.cpp:109:24: error: ‘class moveit::planning_interface::MoveGroupInterface’ has no member named ‘getJointModelGroupNames’
   std::copy(move_group.getJointModelGroupNames().begin(), move_group.getJointModelGroupNames().end(),
                        ^
/home/cc/Documents/ws_moveit/src/moveit_tutorials/doc/move_group_interface/src/move_group_interface_tutorial.cpp:109:70: error: ‘class moveit::planning_interface::MoveGroupInterface’ has no member named ‘getJointModelGroupNames’
   std::copy(move_group.getJointModelGroupNames().begin(), move_group.getJointModelGroupNames().end(),
                                                                      ^
make[2]: *** [doc/interactivity/CMakeFiles/interactivity_utils.dir/src/interactive_robot.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [doc/move_group_interface/CMakeFiles/move_group_interface_tutorial.dir/src/move_group_interface_tutorial.cpp.o] Error 1
make[1]: *** [doc/move_group_interface/CMakeFiles/move_group_interface_tutorial.dir/all] Error 2
make[1]: *** [doc/interactivity/CMakeFiles/interactivity_utils.dir/all] Error 2
make: *** [all] Error 2
cd /home/cc/Documents/ws_moveit/build/moveit_tutorials; catkin build --get-env moveit_tutorials | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << moveit_tutorials:make                [ Exited with code 2 ]      
Failed    <<< moveit_tutorials                     [ 19.2 seconds ]            
[build] Summary: 2 of 3 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 22.0 seconds total.
```

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Binary build (kinetic)

Thank you.

mlautman, 2019-06-25:
The issue is that you are mixing Isometry3d with Affine3d. You should use these tutorials for kinetic: http://docs.ros.org/kinetic/api/moveit_tutorials/html/index.html

For Kinetic, you should not see any instances of Isometry3d in your code base.

Can you confirm that you are indeed using the kinetic-devel branch of moveit_tutorials?

roboticsanalyst, 2019-07-30:
Hey @mlautman, thanks a lot for your suggestion.   I had the same problem when I tried building the  moveit_tutorials from master branch but went away with kinetic-devel b.

srinivasgln, 2019-07-30:
Hey @mlautman , thanks a lot for your suggestion. I had the same issue when I tried building the moveit_tutorials from master branch but went away with kinetic-devel b.

rhaschke, 2019-08-02:
Closing, as this is a compatibility issue as described by @mlautman.

closed 2019-08-02 by rhaschke