moveit/moveit_tutorials #342: std::bad_alloc when calling addCollisionObjects after migrating to melodic

GunnarMaer, 2019-06-20:
Hello everyone,

I slightly modified [the tutorial, the part where collision objects are added](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/move_group_interface/move_group_interface_tutorial.html#adding-removing-objects-and-attaching-detaching-objects )(just changed the size and position of the box).

My node worked perfectly on ROS Kinetic. Now I moved to ROS **Melodic** and I get an error while executing line [325](https://github.com/ros-planning/moveit_tutorials/blob/77e09127ca482bd7cd1b02f7081ac932ac3e3d75/doc/move_group_interface/src/move_group_interface_tutorial.cpp#L325 ): `planning_scene_interface.addCollisionObjects(collision_objects);`

The error looks like this:
```
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
```

For testing purpose I tried to comment line 321 out, i.e. giving the addCollisionObjects an empty vector. In this case no error occurs. Also if I comment out line 325, I get no error. Therefore I guess the vector should be okay, but the function can't handle it somehow.

The rest of my node (400 lines of code) works as intended and as it worked on Kinetic.

Thank you for your insights!

GunnarMaer, 2019-06-21:
If I print out the entry of the vector (`ROS_INFO_STREAM(collision_objects[0]);`), I get the following results:

```
header: 
  seq: 0
  stamp: 0.000000000
  frame_id: world
id: table
type: 
  key: 
  db: 
primitives[]
  primitives[0]: 
    type: 1
    dimensions[]
      dimensions[0]: 3.2
      dimensions[1]: 2
      dimensions[2]: 0.2
primitive_poses[]
  primitive_poses[0]: 
    position: 
      x: 0.6
      y: 0
      z: -0.1
    orientation: 
      x: 0
      y: 0
      z: 0
      w: 1
meshes[]
mesh_poses[]
planes[]
plane_poses[]
operation: 0
```

rhaschke, 2019-06-21:
This bug was fixed in #1438. Unfortunately we didn't prepare a new Melodic release yet.
Please build from source.

closed 2019-06-21 by rhaschke

GunnarMaer, 2019-06-21:
Thanks for your reply. I hoped it would be that easy. But unfortunately, I still get the same error.

Just to be sure I did everything correctly, these are the things I did:
- I backupped an own moveit config package (only the one in the `src` directory)
- I deleted my whole `ws_moveit` folder
- I followed [this tutorial](https://moveit.ros.org/install/source/) to install from source 
- Before calling `catkin build`, I put my own package back into the `src` folder
- I called `catkin build`

rhaschke, 2019-06-21:
Did you resourced your new workspace? Where does `roscd` moveit_core bring you? To your workspace or to `/opt/ros`?

GunnarMaer, 2019-06-24:
Yes I did. It brings me to my MoveIt workspace: `~/ws_moveit/src/moveit/moveit_core`

rhaschke, 2019-06-24:
To further investigate your problem, it would be helpful if you could distill a minimal example that exhibits the failure. Thanks.

GunnarMaer, 2019-06-25:
This might not be the best way to give you my minimal example, if you know anything better, please tell me :)

File `posegiver.cpp`:

```
#include <iostream>
#include <ros/ros.h>

#include <moveit/move_group_interface/move_group_interface.h>
#include <moveit/planning_scene_interface/planning_scene_interface.h>

#include <moveit_msgs/AttachedCollisionObject.h>
#include <moveit_msgs/CollisionObject.h>

using namespace std;

class Testclass{
public:
    Testclass(){
    }

    ~Testclass(){
    }

    void start(){
        moveit::planning_interface::MoveGroupInterface move_group_panda("panda_arm");

        // Table as a collision object
        moveit::planning_interface::PlanningSceneInterface planning_scene_interface;
        moveit_msgs::CollisionObject table;
        table.header.frame_id = move_group_panda.getPlanningFrame();
        table.id = "table";
        shape_msgs::SolidPrimitive primitive;
        primitive.type = primitive.BOX;
        primitive.dimensions.resize(3);
        primitive.dimensions[0] = 3.2;
        primitive.dimensions[1] = 2.0;
        primitive.dimensions[2] = 0.2;
        geometry_msgs::Pose table_pose;
        table_pose.orientation.w = 1.0;
        table_pose.position.x = 0.6;
        table_pose.position.y = 0.0;
        table_pose.position.z = -0.1;
        table.primitives.push_back(primitive);
        table.primitive_poses.push_back(table_pose);
        table.operation = table.ADD;
        std::vector<moveit_msgs::CollisionObject> collision_objects;
        collision_objects.push_back(table);
        ROS_INFO_STREAM("1");
        //std::vector<moveit_msgs::CollisionObject> collision_objects{table};
        planning_scene_interface.addCollisionObjects(collision_objects);
        ROS_INFO_STREAM("2");

    }
};

int main(int argc, char* argv[]) {
    ros::init(argc, argv, "minimalExample");
    ros::AsyncSpinner spinner(4);
    spinner.start();
    Testclass myTestClass;
    myTestClass.start();
    return 0;
}

```

The `CMakeLists.txt` file:

```
cmake_minimum_required(VERSION 2.8)
project(testProject)
add_compile_options(-std=c++11)

find_package(catkin REQUIRED
    COMPONENTS
    geometry_msgs
    roscpp
    moveit_ros_planning_interface
    moveit_visual_tools
)

find_package (Eigen3 REQUIRED)

catkin_package(
        CATKIN_DEPENDS std_msgs roscpp geometry_msgs
)

include_directories(
 include
  ${catkin_INCLUDE_DIRS}
  ${Eigen3_INCLUDE_DIRS}
  roscpp
  std_msgs
)

add_executable(posegiver "posegiver.cpp")

target_link_libraries(posegiver
        ${catkin_LIBRARIES}
)
```

Once you did `catkin_build` (in the catkin_ws), you should be able to see the result by calling:
`roslaunch panda_ur5_moveit_config demo.launch ` in the first terminal.
`rosrun testProject posegiver` in the second terminal.

rhaschke, 2019-06-27:
I was able to reproduce the issue. Will have a deeper look asap.

rhaschke, 2019-06-27:
I guess, you are building moveit_msgs from source (as well as MoveIt from master branch), do you?
The `CollisionObject` msg has changed recently and now causes strange stack errors on my side: the size of the CollisionObject vector is reported to be huge (but only deep in the std library).

rhaschke, 2019-06-27:
As soon as I configured the `moveit_msgs` dependency correctly (and rebuild from scratch, i.e. after catkin clean), everything worked as expected.

GunnarMaer, 2019-06-30:
Great, thank you very much for your effort! 

Yes, I built MoveIt completely from source from the master branch. 
What do you mean by correct dependency? Shall I work with another version? For moveit_msgs itself there is no master branch...

GunnarMaer, 2019-07-15:
Since I needed a quick solution, I did a workaround by adding my collision object directly into the URDF with selfmade `.stl` files.

rhaschke, 2019-07-15:
Sorry, looks like I missed your previous response. 
Please exactly describe your catkin workspace: Which packages are part of your workspace (and with which version)? For me, I think using released MoveIt binaries (not yet using latest melodic-devel of moveit_msgs) together with latest moveit_msgs and the tutorial failed.

The latest binary releases of MoveIt (1.0.1 and 1.0.2) build on moveit_msgs 0.10.0

Robotawi, 2020-06-13:
Hello Dr. @rhaschke 
I am experiencing the same ‘bad_alloc’ problem at the beginning of the [tutorials](https://ros-planning.github.io/moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html). The error occurs when I roslaunch the demo of the panda_moveit_config
```
roslaunch panda_moveit_config demo.launch rviz_tutorial:=true
```
I am using ROS melodic on Ubuntu 18.04 

I tried to solve it, base on the previous comment about movit_msgs but so far my trials are not very successful. moveit_msgs pkg is installed with moveit at the beginning using
```
wstool merge -t . https://raw.githubusercontent.com/ros-planning/moveit/master/moveit.rosinstall
```
I removed this moveit_msgs pkg, installed it again from the [source](https://github.com/ros-planning/moveit_msgs), then catkin clean and catkin build. This didn’t solve the ‘bad_alloc’ problem.

Would you mind sharing a bit more details about the solution that worked?
Thank you!

rhaschke, 2020-06-13:
Installing all MoveIt packages from recent Melodic binaries, I don't have any issues running  panda_moveit_config's demo.launch. Which binary causes the problem? On which action, just starting demo.launch?
Maybe, try to update your debian packages and run against /opt/ros first.

Robotawi, 2020-06-13:
Thanks for your reply. 
I encountered this problem while working on the [master branch tutorials](https://ros-planning.github.io/moveit_tutorials). The error happens when running the demo.launch file.
___
Based on your reply, I switched to the [melodic branch tutorials](http://docs.ros.org/melodic/api/moveit_tutorials/html/index.html). **The demo.launch file works well**, but there is a similar `bad_alloc` error when I try to roslaunch the move_group_interface_tutorial in the [Move Group C++ interface](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/move_group_interface/move_group_interface_tutorial.html) tutorial

This is shown
`
[ INFO] [1592086282.256616952]: Loading robot model 'panda'...
`
Then the error:
`
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
`

The command that causes the error: 
```
roslaunch moveit_tutorials move_group_interface_tutorial.launch
``` 
I would appreciate your insight into this issue.
Thank you!

rhaschke, 2020-06-14:
Sorry, I can't reproduce your issue. I have build against MoveIt binary packages as follows:
```
mkdir -p /tmp/ros/src
cd /tmp/ros/src
git clone -b melodic-devel https://github.com/ros-planning/moveit_tutorials
cd ..
catkin config --extend /opt/ros/melodic --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo
catkin build

source devel/setup.bash
roslaunch panda_moveit_config demo.launch
roslaunch moveit_tutorials move_group_interface_tutorial.launch
```
Usually, such bad_alloc errors indicate some fundamental issues in building/linking, typically because you mix headers and libs of incompatible versions. Try to build from a clean workspace as indicate above.

Robotawi, 2020-06-15:
I am truly thankful for your help. I tried another machine with a fresh Ubuntu 18.04 installation and its working well. My initial guess is that there is a compatibility issue with the boost library version I had on my computer from before. I will confirm once I become sure about the reason.

rhaschke, 2020-06-15:
Yeah, using another boost version for parts of the ROS ecosystem will not work...

nehagarg, 2020-09-30:
I did the clean build as suggested by https://github.com/ros-planning/moveit_tutorials/issues/342#issuecomment-643735398 and installed pandas_moveit_config fromt apt but I am still getting the bad_alloc error as reported by https://github.com/ros-planning/moveit_tutorials/issues/342#issuecomment-643693131. I am running ros melodic on ubuntu 18. My boost library version is 1.65.1.

 Would appreciate if you could share your insights on what could be the reason for this error. 
Thanks

nehagarg, 2020-09-30:
Figured that cmake was finding an older boost version by running catkin build -v. Managed to make it work by changing CMakeLists.txt of moveit_tutorials to find the version 1.65.1 and specifying -DBOOST_LIBRARYDIR=/usr/lib/x86_64-linux-gnu/ -DBOOST_INCLUDEDIR=/usr/include in catkin config.

dbdxnuliba, 2022-05-06:
thanks a lot @nehagarg  , your solution is very good !

> Figured that cmake was finding an older boost version by running catkin build -v. Managed to make it work by changing CMakeLists.txt of moveit_tutorials to find the version 1.65.1 and specifying -DBOOST_LIBRARYDIR=/usr/lib/x86_64-linux-gnu/ -DBOOST_INCLUDEDIR=/usr/include in catkin config.

xiaopgaxiaopg, 2022-05-25:
> 
I have the same problem, but I don't know how to modify it, can I communicate it