moveit/moveit_tutorials #354: Broken links in Indigo moveit_tutorials

rhaschke, 2019-06-28:
As [Travis reveals](url), there are several broken links to http://picknik.io/moveit_wiki in the Indigo branch of the tutorials. @davetcoleman introduced this site in #17 but it was obviously removed in the meantime, probably in conjunction with #142. A corresponding fix should be applied to the Indigo branch as well.

```
./doc/ros_visualization/visualization_tutorial.rst:177:  <http://picknik.io/moveit_wiki/index.php?title=PR2/Gazebo/Quick_Start>`_ - You can now
./doc/setup_assistant/setup_assistant_tutorial.rst:12:can go through the `SRDF Overview <http://picknik.io/moveit_wiki/index.php?title=SRDF>`_
./doc/setup_assistant/setup_assistant_tutorial.rst:292:* See the `SRDF <http://picknik.io/moveit_wiki/index.php?title=SRDF>`_ page for more
./doc/setup_assistant/setup_assistant_tutorial.rst:300:  documentation <http://picknik.io/moveit_wiki/index.php?title=URDF>`_ to see how the
```

davetcoleman, 2019-06-28:
I fixed this in later versions of the tutorials (kinetic?), and don't have bandwidth to fix an EOL version of the tutorials. We shouldn't waste time on this.

rhaschke, 2019-06-29:
Your decision.

closed 2019-06-29 by rhaschke