moveit/moveit_tutorials #359: error at openrave "$MYROBOT_NAME".dae

lucafei, 2019-07-12:
### Description

Overview of your issue here.
hello，guys， i want to use ikfast in moveit and so i followed the moveit tutorial， but when i come to the step：openrave "$MYROBOT_NAME".dae。
it has a error in my console：
[plugindatabase.h:645 Create] Failed to create name fcl_, interface collisionchecker 
[colladareader.cpp:2235 FillGeometryColor] transparecy is 1.000000, which means the item will be rendered invisible, this must be a mistake so setting to opaque (1) 
[colladareader.cpp:2235 FillGeometryColor] transparecy is 1.000000, which means the item will be rendered invisible, this must be a mistake so setting to opaque (1)
i‘m now crazy and dont know what caused this, can you help me?
### Your environment
* ROS Distro: [Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.
i followed the step to install openrave at the webseit http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ikfast/ikfast_tutorial.html
### Expected behaviour
Tell us what should happen
when i come to the step：openrave "$MYROBOT_NAME".dae it happened the above error
### Backtrace or Console output
[plugindatabase.h:645 Create] Failed to create name fcl_, interface collisionchecker 
[colladareader.cpp:2235 FillGeometryColor] transparecy is 1.000000, which means the item will be rendered invisible, this must be a mistake so setting to opaque (1) 
[colladareader.cpp:2235 FillGeometryColor] transparecy is 1.000000, which means the item will be rendered invisible, this must be a mistake so setting to opaque (1)
Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

gavanderhoorn, 2019-07-12:
Cross-post of [error at openrave "$MYROBOT_NAME".dae](https://answers.ros.org/question/328357) on ROS Answers.

rhaschke, 2019-07-12:
According to the error message your robot has fully transparent parts and you should fix this.
No idea though for the fcl error. I would be helpful if you really would provide your urdf (e.g. in a gist) so that we can reproduce the problem.

mlautman, 2019-07-12:
That's not really an error but more of a warning. You should be able to continue past that warning

closed 2019-07-12 by mlautman

IndustrialEngStudent, 2019-07-24:
@lucafei  did you find out what was wrong? I have the same problem

lucafei, 2019-08-12:
no,i didnt find a solution and i have already to change the precision but it doesnt work, @IndustrialEngStudent did you find the solution for this?

rhaschke, 2019-08-12:
I updated the IKFast tutorial a while ago (#332, unfortunately its approval is still pending). 
Could you, please, try this newer version linked [here](https://github.com/ros-planning/moveit_tutorials/blob/d6301ce5b40cfa497ee5a79b2096f4cd92d79ae5/doc/ikfast/ikfast_tutorial.rst)?
Given your panda_arm.urdf, I had no issues to generate the IKFast plugin following this method:
```sh
rosrun moveit_kinematics auto_create_ikfast_moveit_plugin.sh --iktype Transform6D panda_arm.urdf panda_arm panda_link0 panda_link8
```

ruiaisun, 2021-11-09:
Hello, Can you tell me how to fix the transparency problem here? I tried to google the solution but did not find anyone. @rhaschke

rhaschke, 2021-11-09:
According to https://github.com/ros-planning/moveit_tutorials/issues/359#issuecomment-511008617, the transparency message can be considered as a warning. It doesn't affect the outcome of the IKfast generator.
If you want to fix the warning, you should have a look into your .dae file and fix the transparency of some link therein.