moveit/moveit_tutorials #363: moveit_tutorials catkin build failded

Reggelord, 2019-07-30:
### Description
Hi everyone

I was following tutorial for ROS [melodic ](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html) but failed to build workspace with `catkin build`

Anyone knows how to fix it ?
### Your environment

* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary build (melodic)

### Backtrace or Console output

````
----------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/melodic
Workspace:                   /home/b/ws_moveit
----------------------------------------------------------
Build Space:        [exists] /home/tom/ws_moveit/build
Devel Space:        [exists] /home/tom/ws_moveit/devel
Install Space:      [unused] /home/tom/ws_moveit/install
Log Space:          [exists] /home/tom/ws_moveit/logs
Source Space:       [exists] /home/tom/ws_moveit/src
DESTDIR:            [unused] None
----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
----------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
----------------------------------------------------------
Workspace configuration appears valid.
----------------------------------------------------------
[build] Found '2' packages in 0.0 seconds.                                                             
[build] Package table is up to date.                                                                   
Starting  >>> panda_moveit_config                                                                      
Finished  <<< panda_moveit_config                [ 0.2 seconds ]                                       
Starting  >>> moveit_tutorials                                                                         
_______________________________________________________________________________________________________
Errors     << moveit_tutorials:make /home/tom/ws_moveit/logs/moveit_tutorials/build.make.003.log     
In file included from /opt/ros/melodic/include/moveit/collision_detection_fcl/collision_robot_fcl.h:40:0,
                 from /opt/ros/melodic/include/moveit/collision_detection_fcl/collision_world_fcl.h:40,
                 from /home/tom/ws_moveit/src/moveit_tutorials/doc/visualizing_collisions/src/visualizing_collisions_tutorial.cpp:47:
/opt/ros/melodic/include/moveit/collision_detection_fcl/collision_common.h: In constructor ‘collision_detection::FCLGeometry::FCLGeometry(fcl::CollisionGeometryd*, const moveit::core::LinkModel*, int)’:
/opt/ros/melodic/include/moveit/collision_detection_fcl/collision_common.h:191:24: error: invalid use of incomplete type ‘using element_type = class fcl::CollisionGeometry {aka class fcl::CollisionGeometry}’
     collision_geometry_->setUserData(collision_geometry_data_.get());
                        ^~
In file included from /opt/ros/melodic/include/moveit/collision_detection_fcl/collision_common.h:43:0,
                 from /opt/ros/melodic/include/moveit/collision_detection_fcl/collision_robot_fcl.h:40,
                 from /opt/ros/melodic/include/moveit/collision_detection_fcl/collision_world_fcl.h:40,
                 from /home/tom/ws_moveit/src/moveit_tutorials/doc/visualizing_collisions/src/visualizing_collisions_tutorial.cpp:47:
/opt/ros/melodic/include/moveit/collision_detection_fcl/fcl_compat.h:48:7: note: forward declaration of ‘using element_type = class fcl::CollisionGeometry {aka class fcl::CollisionGeometry}’
 class CollisionGeometry;
       ^~~~~~~~~~~~~~~~~
In file included from /opt/ros/melodic/include/moveit/collision_detection_fcl/collision_robot_fcl.h:40:0,
                 from /opt/ros/melodic/include/moveit/collision_detection_fcl/collision_world_fcl.h:40,
                 from /home/tom/ws_moveit/src/moveit_tutorials/doc/visualizing_collisions/src/visualizing_collisions_tutorial.cpp:47:
/opt/ros/melodic/include/moveit/collision_detection_fcl/collision_common.h: In constructor ‘collision_detection::FCLGeometry::FCLGeometry(fcl::CollisionGeometryd*, const moveit::core::AttachedBody*, int)’:
/opt/ros/melodic/include/moveit/collision_detection_fcl/collision_common.h:197:24: error: invalid use of incomplete type ‘using element_type = class fcl::CollisionGeometry {aka class fcl::CollisionGeometry}’
     collision_geometry_->setUserData(collision_geometry_data_.get());
                        ^~
In file included from /opt/ros/melodic/include/moveit/collision_detection_fcl/collision_common.h:43:0,
[... 303 lines cut ...]
                 from /usr/local/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/melodic/include/ros/time.h:58,
                 from /opt/ros/melodic/include/ros/ros.h:38,
                 from /home/tom/ws_moveit/src/moveit_tutorials/doc/visualizing_collisions/src/visualizing_collisions_tutorial.cpp:39:
/usr/include/c++/7/bits/shared_ptr_base.h: In instantiation of ‘std::__shared_ptr<_Tp, _Lp>::__shared_ptr(_Yp*) [with _Yp = fcl::CollisionGeometry; <template-parameter-2-2> = void; _Tp = fcl::CollisionGeometry; __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)2]’:
/usr/include/c++/7/bits/shared_ptr.h:129:46:   required from ‘std::shared_ptr<_Tp>::shared_ptr(_Yp*) [with _Yp = fcl::CollisionGeometry; <template-parameter-2-2> = void; _Tp = fcl::CollisionGeometry]’
/opt/ros/melodic/include/moveit/collision_detection_fcl/collision_common.h:189:117:   required from here
/usr/include/c++/7/bits/shared_ptr_base.h:1082:25: error: invalid application of ‘sizeof’ to incomplete type ‘fcl::CollisionGeometry’
    static_assert( sizeof(_Yp) > 0, "incomplete type" );
                         ^
make[2]: *** [doc/visualizing_collisions/CMakeFiles/visualizing_collisions_tutorial.dir/src/visualizing_collisions_tutorial.cpp.o] Error 1
make[1]: *** [doc/visualizing_collisions/CMakeFiles/visualizing_collisions_tutorial.dir/all] Error 2
make: *** [all] Error 2
cd /home/tom/ws_moveit/build/moveit_tutorials; catkin build --get-env moveit_tutorials | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.......................................................................................................
Failed     << moveit_tutorials:make              [ Exited with code 2 ]                                
Failed    <<< moveit_tutorials                   [ 10.3 seconds ]                                      
[build] Summary: 1 of 2 packages succeeded.                                                            
[build]   Ignored:   None.                                                                             
[build]   Warnings:  None.                                                                             
[build]   Abandoned: None.                                                                             
[build]   Failed:    1 packages failed.                                                                
[build] Runtime: 10.6 seconds total.    

```

charvi-077, 2019-10-02:
I am also facing the same problem for moveit_tutorials package , also after catkin build my laptop freezes .
Ros distro : melodic
OS version: ubuntu 18.04.

mlautman, 2019-10-09:
First off, I would recommend building off the master branch unless you have a strong reason not to. Virtually all of the maintainers work off of master so while melodic-devel would ideally be more stable,  in practice master seems to be the right choice. 

The link you sent over has the instructions for building `master` not `melodic-devel`, was that your intention? Looking at the error, it seems you are building off of the system install of MoveIt `/opt/ros/melodic/include/moveit`. This works fine on my machine for the `melodic-devel` branch of the tutorials but will NOT work for the `master` branch of the tutorials. 

For the `melodic-devel` branch of the tutorials, be sure to follow the instructions [linked to here](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/getting_started/getting_started.html). If you are still having issues run `sudo apt update; sudo apt upgrade` to see if the most up to date debians fix your issue. 

For the master branch, clean out your workspace completely (`cd ~/ws_moveit/; catkin clean -y; rm -rf .catkin_tools`) and make sure you follow all the instructions in Getting Started [linked to here](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html). **Do not skip this step**:
`wstool merge -t . https://raw.githubusercontent.com/ros-planning/moveit/master/moveit.rosinstall` which downloads the master branch of MoveIt and it's dependencies to your workspace. You should then be able to build from source just fine. 
 
I am going to close this issue since I cannot reproduce it locally. If you follow the instructions above and are still having issues, feel free to re-open this issue.

happy motion planning!

closed 2019-10-09 by mlautman