moveit/moveit_tutorials #383: Solution for Panda MoveIt 3d Model

Greg9914, 2019-08-22:
### Description

Hello,
I have followed all the instruction procedures from http://docs.ros.org/melodic/api/moveit_tutorials/html/index.html
But, after my installation, as I opened my RViz with MoveIt, the robot model in 3d showed some errors (like if it was compacted / static). 

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04.3

### Steps to reproduce
The solution was to install all Franka packages:
cd ~/catkin_ws/src
sudo apt-get install ros-melodic-install*

And after the installation:
catkin_make

Please make sure that franka_ros package is located in /catkin_ws/src.

### Expected behaviour
After the procedure, you should close the terminal and reopen, and command:
roslaunch panda_moveit_config demo.launch rviz_tutorial:=true
The Panda Model should look normal now.

rhaschke, 2019-08-23:
Are you sure you exactly followed the instructions in [Getting Started](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/getting_started/getting_started.html#build-your-catkin-workspace)?
Particularly, I think you missed to install package dependencies like this:
`rosdep install -y --from-paths . --ignore-src --rosdistro melodic`

I just followed these instructions in a clean docker container and everything worked perfectly.

closed 2019-08-23 by rhaschke

gavanderhoorn, 2019-08-23:
@Greg9914: you may be running into a known problem with `urdfdom`. See [Rviz shows robot in a "crippled way" using MotionPlanning](https://answers.ros.org/question/319077) on ROS Answers for an example Q&A about this.

Without a screenshot it's hard to know what "compacted" really means though.

@rhaschke: this particular problem depends on locale settings, so the fact that it works in a Docker container for you (that is most likely using `en_US.UTF8`) isn't very conclusive evidence everything is fine ;) (although it also isn't necessarily something we can do something about, other than suggest the work-around).

rhaschke, 2019-08-23:
@gavanderhoorn, as @Greg9914  mentioned that after installing all dependencies everything worked fine for him, I think he just didn't exactly follow the instructions.

gavanderhoorn, 2019-08-23:
O wait. You're correct. My apologies.

I did not expect an issue with a 'solution' in the issue description.

`/ignore`.