moveit/moveit_tutorials #385: Adding new .rst pages to the tutorials makes Travis fail

rhaschke, 2019-08-23:
Due to some hard-coded links to github ("Edit on GitHub" and "Open PR on this GitHub page"), Travis will always complain on newly created tutorials, because these links don't yet exist on github.
An example can be found [here](https://travis-ci.org/ros-planning/moveit_tutorials/builds/575909907).

v4hn, 2019-10-17:
The same is true for other files referenced that way, whenever the new content is directly linked.

A possible solution I could imagine would rewrite all URLs during CI to reference the files in the pull-request branch instead.

rhaschke, 2019-10-17:
There is an option `--url-swap` of `htmlproofer`, which should translate links to local links for checking. Will give it a try...

rhaschke, 2019-10-17:
Fixed via #414.

closed 2019-10-17 by rhaschke