moveit/moveit_tutorials #398: moveit_tutorials catkin build failed

surfii3z, 2019-09-20:
### Description
I git clone moveit_tutorials and panda_moveit_config in to ~/catkin_ws/src and try to catkin build following this link http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/getting_started/getting_started.html.

However it failed with this error

Errors     << moveit_tutorials:make /home/jedsadakorn/catkin_ws/logs/moveit_tutorials/build.make.001.log                                                                                                   
In file included from /usr/include/eigen3/Eigen/Core:297:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model_group.h:41,
                 from /opt/ros/kinetic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model_loader/robot_model_loader.h:41,
                 from /home/jedsadakorn/catkin_ws/src/moveit_tutorials/doc/robot_model_and_robot_state/src/robot_model_and_robot_state_tutorial.cpp:40:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/jedsadakorn/catkin_ws/src/moveit_tutorials/doc/robot_model_and_robot_state/src/robot_model_and_robot_state_tutorial.cpp:118:102:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:330:5: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
In file included from /usr/include/eigen3/Eigen/Core:297:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model_group.h:41,
                 from /opt/ros/kinetic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model_loader/robot_model_loader.h:41,
                 from /home/jedsadakorn/catkin_ws/src/moveit_tutorials/doc/interactivity/include/interactivity/interactive_robot.h:43,
                 from /home/jedsadakorn/catkin_ws/src/moveit_tutorials/doc/interactivity/src/interactive_robot.cpp:39:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/jedsadakorn/catkin_ws/src/moveit_tutorials/doc/interactivity/src/interactive_robot.cpp:93:32:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:330:5: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
In file included from /usr/include/eigen3/Eigen/Core:297:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model/joint_model_group.h:41,
                 from /opt/ros/kinetic/include/moveit/robot_model/robot_model.h:47,
                 from /opt/ros/kinetic/include/moveit/robot_model_loader/robot_model_loader.h:41,
                 from /home/jedsadakorn/catkin_ws/src/moveit_tutorials/doc/state_display/src/state_display_tutorial.cpp:40:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/home/jedsadakorn/catkin_ws/src/moveit_tutorials/doc/state_display/src/state_display_tutorial.cpp:101:114:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:330:5: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
make[2]: *** [doc/robot_model_and_robot_state/CMakeFiles/robot_model_and_robot_state_tutorial.dir/src/robot_model_and_robot_state_tutorial.cpp.o] Error 1
make[1]: *** [doc/robot_model_and_robot_state/CMakeFiles/robot_model_and_robot_state_tutorial.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [doc/state_display/CMakeFiles/state_display_tutorial.dir/src/state_display_tutorial.cpp.o] Error 1
make[1]: *** [doc/state_display/CMakeFiles/state_display_tutorial.dir/all] Error 2
make[2]: *** [doc/interactivity/CMakeFiles/interactivity_utils.dir/src/interactive_robot.cpp.o] Error 1
make[1]: *** [doc/interactivity/CMakeFiles/interactivity_utils.dir/all] Error 2
make: *** [all] Error 2
cd /home/jedsadakorn/catkin_ws/build/moveit_tutorials; catkin build --get-env moveit_tutorials | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Failed     << moveit_tutorials:make               [ Exited with code 2 ]                                                                                                                                   
Failed    <<< moveit_tutorials                    [ 14.8 seconds ]                                                                                       

Overview of your issue here.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04

rhaschke, 2019-09-20:
As you are working in a Kinetic environment, you should use the kinetic branch of the tutorials:
http://docs.ros.org/kinetic/api/moveit_tutorials/html/index.html

closed 2019-09-20 by rhaschke

surfii3z, 2019-09-20:
Thank you for your help