moveit/moveit_tutorials #411: Move Group C++ interface does not tell you how to successfully use MoveGroupInterface

robottoearth, 2019-10-10:
### Description

I needed to use computeCartesianPath() and started looking at the tutorials to go about it. If only robot_description (with the xacro) is launched, move_group_interface_tutorial.cpp (https://github.com/ros-planning/moveit_tutorials/blob/master/doc/move_group_interface/src/move_group_interface_tutorial.cpp) **will just halt indefinitely at initializing move group** (https://github.com/ros-planning/moveit_tutorials/blob/master/doc/move_group_interface/src/move_group_interface_tutorial.cpp#L67).

This is because move_group.launch needs to be launched (https://github.com/ros-planning/panda_moveit_config/blob/melodic-devel/launch/move_group.launch), which I figured from demo.launch.

It would be great to have some documentation on how to use group outside of this demo and what all arguments move group needs to run. For now I would have to look up a lot of code to figure out what I need for using computeCartesianPath. This involves changing CMakelists and package.xml as well. If documentation is updated to reflect these 3 things I believe moveit would be much easier to work with.

Overview of your issue here.

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04
* Source or Binary build?
* If binary, which release version? 
ros-melodic-moveit:
  Installed: 1.0.1-0bionic.20190320.192509
  Candidate: 1.0.1-0bionic.20190320.192509
  Version table:
 *** 1.0.1-0bionic.20190320.192509 500
        500 http://packages.ros.org/ros/ubuntu bionic/main amd64 Packages
        100 /var/lib/dpkg/status

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Backtrace or Console output
`[ INFO] [1570719154.144350380]: Loading robot model 'visual'...
[ WARN] [1570719154.342459324]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/Lowry/kinematics_solver_attempts' from your configuration.
`
Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

truhoang-ubtrobot, 2019-11-18:
@robottoearth, you need move_group.launch to setup the topics and services called within computeCartesianPath(). From my understanding of the code, this function is a wrapper that calls actions and services from the core MoveIt! API.

Briancbn, 2019-11-20:
@robottoearth currently you can use the [moveit_setup_assistant](https://ros-planning.github.io/moveit_tutorials/doc/setup_assistant/setup_assistant_tutorial.html) to generate a demo launch for a custom made robot as well as edit the configuration. 
I also agree that there should be a tutorial that
- Explain the parameters for move_group (usually configured automatically in demo.launch, move_group.launch etc. maybe better comments in the launch file instead?)
- Explain the necessary parameters needed for move_group interface (pose_target, joint_target and Cartesian path) to work.
- What is the 3rd point that you suggested?

truhoang-ubtrobot, 2019-11-20:
@Briancbn 
Have you looked at the tutorial here? [move group interface tutorial](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/move_group_interface/move_group_interface_tutorial.html)

1. Also why would you want to change the move_group.launch? It launches other services that move group interface with.
2. You can have a look at the API here: [move group API](http://docs.ros.org/melodic/api/moveit_ros_planning_interface/html/classmoveit_1_1planning__interface_1_1MoveGroupInterface.html)

Briancbn, 2019-11-21:
@truhoang-ubtrobot Thank you for your prompt reply. Yes I have tried the c++ move group interface tutorial. 
- Normally I wouldn't change that. Unless when I tried to tweak the parameters manually. Or using launch files from a external source.
- move group API doesn't tell you each parameter's function, or which topic, action move group calls. As I have used other ROS packages like [move_base](http://wiki.ros.org/move_base), for which the doc is hosted on the wiki, there is always a list with explanation for the parameters, topics and actions that is called, which is helpful when I am reading the launch file or writing my own launch file to interface modules self developed with move_group. For move_group is always here and there in the rest of the tutorial or just searching the source code or using rqt for answers. Not sure if this will be helpful for other developers tho.

truhoang-ubtrobot, 2019-11-21:
@Briancbn I see, I forgot how detailed some packages documentation can be.
My experiences with MoveIt has been a mixed bag too. I generally rely on reading the API and the source to understand how to use it.

I do agree a better documentation would be helpful but I doubt any one would take it on at this point as the developers are working on developing MoveIt for ros2 and the code has become quite complex.