moveit/moveit_tutorials #415: Japanese Translation

RyodoTanaka, 2019-10-25:
### Description
Hi all.
I would like to make Japanese translation page.
I already forked this repository on following link.
https://github.com/tork-a/moveit_tutorials
We will create "ja" directory.
(The word "ja", we respect to [ISO 2 letter Language Codes Rule](https://www.sitepoint.com/iso-2-letter-language-codes/).)
And under the "ja" directory, we will create same file structure as this repository does.
We start translation from "melodic-devel" branch.
For the notification, when you join this translation, I created the Issue on following link.
https://github.com/tork-a/moveit_tutorials/issues/1
If you have interests, questions, or advices be free to comment here or contact me.

Bests,

davetcoleman, 2019-10-25:
As we discussed via email, this sounds like a good plan to me.

The only change I would recommend is to convert the 'master' branch instead? It has a lot of new content that we do not plan on backporting to melodic. Additionally, changes to the master branch get released within minutes, whereas changes to melodic go through the ROS buildfarm and take a few days.

rhaschke, 2019-10-25:
> The only change I would recommend is to convert the 'master' branch instead?

I support this!

> whereas changes to melodic go through the ROS buildfarm and take a few days

a few hours in more correct, I think. But, you are right, it's definitely slower.

RyodoTanaka, 2019-10-26:
@davetcoleman 
Thank you for you support.
> >The only change I would recommend is to convert the 'master' branch instead?

> I support this !

I got it. We will start from `master` branch !

RyodoTanaka, 2019-10-28:
Dear All
I am building TravisCI settings.
But it fails with time limit.
https://travis-ci.org/RyodoTanaka/moveit_tutorials
The error occurs on `htmlproofer` part.
I have no idea to fix this.
Any one help me to deal with ? 

Thank you forward.
Bests,

rhaschke, 2019-10-28:
Did you already try to retrigger Travis? Sometimes that already helps.
I don't remember that we had issues with htmlproofer in the past.

RyodoTanaka, 2019-10-28:
@rhaschke 
Thank you for your advice.
I tried to restart the TravisCI.
But I still got the same error.
I tried this `htmlproof` part on my local environment with `time` command.
And it takes around 13 minutes...

``` log
real	13m26.189s
user	13m6.520s
sys	0m3.856s
```

Do you have any idea to make short this time ? 
Bests,

#### Local Environment
- OS : Ubuntu 18.04 LTS
- Kernel : 5.0.0-32-generic #34
- CPU : Intel® Core™ i7-7700HQ CPU @ 2.80GHz × 8
- Memory : 15.5 GiB

rhaschke, 2019-10-28:
By copying everything to a ja subfolder, you essentially doubled the checking effort and time, which is now exceeding the limit.
You should follow the instructions of the link provided with the error:
https://docs.travis-ci.com/user/common-build-problems/#build-times-out-because-no-output-was-received

RyodoTanaka, 2019-10-28:
@rhaschke 
Thanks for your advice, I fixed the timeout error.
https://travis-ci.org/RyodoTanaka/moveit_tutorials/jobs/603820038#L1724

I have other problems with sphinx-build.
But anyway, this problem is solved.
Many thanks !

v4hn, 2019-10-28:
Translated tutorials are of course a wonderful project to reach out to more researchers. :fireworks:

We need to handle updates in the future somehow though.
Did anyone already discuss how to manage changes after the initial translation?
We regularly extend or modify the tutorials (also to account for API changes)
and by far most authors will not adapt the japanese translations.
Also, I believe nobody has the resources to sync the japanese translations with new patches every month.

rhaschke, 2019-10-28:
Even if there are no regular updates to the translations, they are a great contribution.
But you are right, at some point, they will become outdated and require an overhaul.
But I would argue that once a year might be sufficient.

RyodoTanaka, 2019-10-28:
@v4hn @rhaschke 
Thank you for the comment.

> Did anyone already discuss how to manage changes after the initial translation?
I think there are two points to discuss about managing.
1. Technical managing
On the sphinx, there are a support for multi language translation.
https://www.sphinx-doc.org/en/master/usage/advanced/intl.html
But, to utilize this, we need typical build system using `Makefile`.
On this point, I discussed with @davetcoleman on email.
And we thought simply add "ja" directory is the best.
Because, to utilize them takes costs.
I will be grad, if any one has better idea and tell me.

2. Maintenance the Japanese translation
As @v4hn said, I also afraid for that.
We just start to translate, and do not have so many people to maintenance the translation repo.
I would like to have an effort to keep updating, but I could not promise to keep updating for every month.
About the cycle of updating, as @rhaschke said, once on year is possible for us !
And I think to set a issue updating the translation on the WMD in Japan would be good.

gavanderhoorn, 2019-10-28:
>     2\. About the cycle of updating, as @rhaschke said, once on year is possible for us !
>         And I think to set a issue updating the translation on the WMD in Japan would be good.

would it be an idea to make this cycle explicit somewhere?

For instance: add a notice at the top of every translated page, stating that it's only updated once a year (or whatever the period ends up being) with a prominent link to the English version of the tutorials.

RyodoTanaka, 2019-10-28:
@gavanderhoorn 
> would it be an idea to make this cycle explicit somewhere?

I did not have detail idea.
But it sounds good.

For my idea, if some one want to update with not synchronized timing, to merge that would be good.
And the idea to set the issue updating once a year guarantees the minimum update frequency.

RyodoTanaka, 2019-10-29:
Finally, I divide testing about default directory and "ja" directory.
And succeed !
https://travis-ci.org/tork-a/moveit_tutorials

Thank you for the help.