moveit/moveit_tutorials #417: Segmentation fault (core dumped) when creating cpp file with IKFast

arrijon96, 2019-10-26:
Hello!

I am trying to get an analytical inverse kinematics solution for my robot. To do so, I followed this tutorial (http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/ikfast/ikfast_tutorial.html). I am using Melodic and my robot (I attach the urdf file) has got 3DOF.

As a starting point, in order to choose the baselink and eelink I run:
- - -
openrave-robot.py jurp1_1.dae --info links
- - -

and I get:
- - -
name index parents

base_link 0
Shoulder_Link 1 base_link
Elbow_Link 2 Shoulder_Link
Wrist_Link 3 Elbow_Link
EE_Link 4 Wrist_Link
- - -
In order to generate the cpp file with the analytical solution, I chose "Translation3D" as the inverse kinematic type. .

To do so, I run this command:
- - -
python openrave-config --python-dir/openravepy/openravepy/ikfast.py --robot=jurp1.dae --iktype=translation3d --baselink=0 --eelink=3 --freeindex=4 --savefile=~/Desktop/ikfast61_jurp_arm.cpp
- - -
I get the following:
- - -
INFO: moved translation [0, 0, 0] to right end
INFO: moved translation [0, 0, 133/1000] to left end
INFO: [[0, 0, 1, 11367/50000],[0, 1, 0, 0],[-1, 0, 0, 13/10]]
INFO: [[cos(j0), -sin(j0), 0, 0],[sin(j0), cos(j0), 0, 0],[0, 0, 1, 0]]
INFO: [[0, 0, 1, 0],[1, 0, 0, 0],[0, 1, 0, 0]]
INFO: [[cos(j1), -sin(j1), 0, 0],[sin(j1), cos(j1), 0, 0],[0, 0, 1, 0]]
INFO: [[0, 0, -1, 0],[-1, 0, 0, 0],[0, 1, 0, 227/1000]]
INFO: [[cos(j2), -sin(j2), 0, 0],[sin(j2), cos(j2), 0, 0],[0, 0, 1, 0]]
INFO: [[0, 0, 1, 0],[1, 0, 0, 0],[0, 1, 0, 0]]
INFO: ikfast translation3d: [j0, j1, j2]
INFO: depth=0 c=1, [] [j0, j1, j2]: cases=None
INFO: j0 solution: equations used for atan2: [pxsin(j0) - pycos(j0), -pxcos(j0) - pysin(j0) + 227/1000]
INFO: adding atan2(227py, 227px) = 227Abs(px) + 227Abs(py) all zeros check
INFO: adding atan2(227py, 227px) = 227Abs(px) + 227Abs(py) all zeros check
INFO: depth=0 c=3, [j0] [j1, j2]: cases=set([])
INFO: adding atan2(pz, -cj0py + pxsj0) = Abs(pz) + Abs(-cj0py + pxsj0) all zeros check
INFO: adding atan2(pz, -cj0py + pxsj0) = Abs(pz) + Abs(-cj0py + pxsj0) all zeros check
INFO: adding atan2(cj0py - pxsj0, pz) = Abs(pz) + Abs(cj0py - pxsj0) all zeros check
INFO: adding atan2(cj0py - pxsj0, pz) = Abs(pz) + Abs(cj0py - pxsj0) all zeros check
INFO: adding atan2(pz, -cj0py + pxsj0) = Abs(pz) + Abs(-cj0py + pxsj0) all zeros check
INFO: adding atan2(pz, -cj0py + pxsj0) = Abs(pz) + Abs(-cj0py + pxsj0) all zeros check
INFO: adding atan2(cj0py - pxsj0, pz) = Abs(pz) + Abs(cj0py - pxsj0) all zeros check
INFO: adding atan2(cj0py - pxsj0, pz) = Abs(pz) + Abs(cj0py - pxsj0) all zeros check
INFO: depth=0 c=5, [j0, j1] [j2]: cases=set([])
INFO: trying to guess variable from [j2]
INFO: have only one variable left j2 and most likely it is not in equations [-pz, pxsin(j0) - pycos(j0), -pxcos(j0) - pysin(j0) + 227/1000, -px2 - py2 - pz**2 + 51529/1000000, pxsin(j0)sin(j1) - pysin(j1)cos(j0) + pzcos(j1), -pxsin(j0)cos(j1) + pycos(j0)cos(j1) + pzsin(j1)]
INFO: depth=0, c=2, iter=0/2, starting newcases: set([py, px])
INFO: depth=2 c=6, [] [j0, j1, j2]: cases=set([py, px])
INFO: depth=0, c=2, iter=1/2, starting newcases: set([Abs(px) + Abs(py)])
INFO: depth=1 c=7, [] [j0, j1, j2]: cases=set([Abs(px) + Abs(py)])
INFO: generating cpp code...
INFO: c=1 var=j0
INFO: c=2 var=j1
INFO: c=3 var=j2
INFO: c=4, store solution
INFO: c=5 var=j1
INFO: c=6 var=j2
INFO: c=7, store solution
Segmentation fault (core dumped)
- - -
Even if it says "INFO: generating cpp code...". The cpp file is never generated. Firstly, I have decreased the amount of decimals in the dae file, so that it would be easier for IKfast to solve. Unfortunately, it did not work. As a last effort I have played around with the inverse kinematics type and baselink/eelink/freeindex numbers (also with no freeindex at all), but I always end up with "Segmentation fault (core dumped)" message.

If someone could help me, I will be grateful.

Jon

[jurp1.zip](https://github.com/ros-planning/moveit_tutorials/files/3775028/jurp1.zip)

Michaelyangyu, 2020-08-03:
I also encountered this problem, have you fixed it?

yifan-hou, 2020-08-13:
I met the same problem. I got the cpp anyway by modifying the last few lines in ikfast.py.

The last lines of ikfast.py looks like
```python
    solvefn=IKFastSolver.GetSolvers()[options.iktype.lower()]
    if options.robot is not None:
        try:
            env=openravepy.Environment()
            kinbody=env.ReadRobotXMLFile(options.robot)
            env.Add(kinbody)
            solver = IKFastSolver(kinbody,kinbody)
            solver.maxcasedepth = options.maxcasedepth
            chaintree = solver.generateIkSolver(options.baselink,options.eelink,options.freeindices,solvefn=solvefn)
            code=solver.writeIkSolver(chaintree,lang=options.lang)
        finally:
            openravepy.RaveDestroy()

    if len(code) > 0:
        with open(options.savefile,'w') as f:
            f.write(code)
```
BTW the file is located at `/usr/local/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/ikfast.py' on my machine.
In my case the segfault happens in ```openravepy.RaveDestroy()```, at which time the ik code is already computed. So I just bypassed the faulty line:

```python
    solvefn=IKFastSolver.GetSolvers()[options.iktype.lower()]
    if options.robot is not None:
        try:
            env=openravepy.Environment()
            kinbody=env.ReadRobotXMLFile(options.robot)
            env.Add(kinbody)
            solver = IKFastSolver(kinbody,kinbody)
            solver.maxcasedepth = options.maxcasedepth
            chaintree = solver.generateIkSolver(options.baselink,options.eelink,options.freeindices,solvefn=solvefn)
            code=solver.writeIkSolver(chaintree,lang=options.lang)
            if len(code) > 0:
                with open(options.savefile,'w') as f:
                    f.write(code)
        finally:
            openravepy.RaveDestroy()
```
Then run ikfast again. It should proceed to write the cpp file.

mohdwaseem27, 2020-08-13:
> I met the same problem. I got the cpp anyway by modifying the last few lines in ikfast.py.
>
> The last lines of ikfast.py looks like
>
> ```python [...]

I did those changes and but got the same segfault

yifan-hou, 2020-08-13:
Yeah but you should have the cpp now.

mohdwaseem27, 2020-08-13:
@yifan-hou yeah I got cpp code, sorry I didn't checked. does this cpp work fine?

thank you

yifan-hou, 2020-08-13:
@mohdwaseem27  No problem.

I can compile and run this file with no error. I need more time to check if the results are correct, which I think is very likely.

ChildofUngolianth, 2021-02-24:
For me reinstalling openrave AND changing the lst lines of ikfast.py worked fine
For Ubuntu 18.04 and Melodic:
Faulty tries:
Installing OpenRave for ikfast package: FIRST TRY, DID NOT WORK: (segementation fault and no cpp could be created from dae)
		$ sudo add-apt-repository ppa:openrave/release
		$ sudo apt-get update
		$ sudo apt-get install openrave0.8-dp-ikfast
	--> Error message, so I changed to second try:
	In catkin_workspace:
		$ git clone https://github.com/ros-planning/moveit_ikfast.git
		$ cd openrave-installation
		$ ./install-dependencies.sh
		$ ./install-osg.sh
		$ ./install-fcl.sh
		$ ./install-openrave.sh

Still did not work

**THIRD TRY, DID WORK** (still throws segemantation fault, but cpp IS created): 
	https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-openrave.md#install-openrave-090-ubuntu-1804-bionic

mlautman, 2021-02-25:
Sounds like you found a workaround. The IKFast solution generator is outside the scope of this repo. For that reason I think it is appropriate to close this issue. If you would recommend a change to the openrave install instructions or add a warning to the tutorials that this is a common failure case please do so in a PR

closed 2021-02-25 by mlautman

1at7, 2022-01-06:
> I did those changes and but got the same segfault
Even after changing the last few lines I was having the same issue. I just rebuild the openrave package following these commands,
`cd openrave/build
cmake .. -DOPT_VIDEORECORDING=OFF  # Avoids AV errors
make -j$(nproc)
sudo make install `

after that I was able to generate the cpp file.

System information:
Ubuntu 18.04
openrave 0.9
ROS melodic
python 2.7