moveit/moveit_tutorials #439: Why given only the quaternion, a pose can be determined

lyh458, 2019-12-21:
### Description
- Files: [MoveIt tutorials, move_group_interface_tutorial.cpp](https://github.com/ros-planning/moveit_tutorials/blob/master/doc/move_group_interface/src/move_group_interface_tutorial.cpp)
- Detailed line: 121, 214, 337. See pose1 defined in that cpp:
```
  geometry_msgs::Pose target_pose1;
  target_pose1.orientation.w = 1.0;
  target_pose1.position.x = 0.28;
  target_pose1.position.y = -0.2;
  target_pose1.position.z = 0.5;
  move_group.setPoseTarget(target_pose1);
```
- Question: 
Why given only the quaternion, a pose can be determined? 
Should not a pose was determined by  position and quaternion? Just as the definition in ``geometry_msgs``
```
MSG: geometry_msgs/Point\n\
# This contains the position of a point in free space\n\
float64 x\n\
float64 y\n\
float64 z\n\
\n\
================================================================================\n\
MSG: geometry_msgs/Quaternion\n\
# This represents an orientation in free space in quaternion form.\n\
\n\
float64 x\n\
float64 y\n\
float64 z\n\
float64 w\n\
```
### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04
* Source

### Steps to reproduce
```
roslaunch panda_moveit_config demo.launch
rosrun moveit_tutorials move_group_interface_tutorial 
```
Thanks!

rhaschke, 2019-12-22:
Note, that all values of messages are initialized to zero. Thus, it suffices to set the w component of the quaternion to one:
https://github.com/ros-planning/moveit_tutorials/blob/7e4004b9f7122722ca913b6206cfac17879881eb/doc/move_group_interface/src/move_group_interface_tutorial.cpp#L121
and subsequently all the position components:
https://github.com/ros-planning/moveit_tutorials/blob/7e4004b9f7122722ca913b6206cfac17879881eb/doc/move_group_interface/src/move_group_interface_tutorial.cpp#L122-L124

closed 2019-12-22 by rhaschke

gavanderhoorn, 2019-12-22:
Cross-posted as well: [why given only the quaternion, a pose can be determined](https://answers.ros.org/question/340487).