moveit/moveit_tutorials #440: pick_place_tutorial fails to run.

JS00000, 2020-01-02:
### Description

I followed the tutorials of MoveIt (latest version, master branch) step by step, everything is ok until I run pick_place_tutorial. The program fails to run, exits in the beginning. 

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source build (following [Source Install](https://moveit.ros.org/install/source/))
* tag: d8678597530a8cbdb4af5075cd230fdaf1af6aae

### Steps to reproduce
I went through the laster version [toturial](https://ros-planning.github.io/moveit_tutorials/index.html) step by step, everything is ok until Pick and Place toturial. Both melodic-devel and kinetic-devel of  panda_moveit_config have been tested, and I got the same result. 

### Backtrace or Console output

<script src="https://gist.github.com/JS00000/09f5d9e7658740acf3f73846102ea4e1.js"></script>

VitaminLL, 2020-05-20:
> ### Description
> I followed the tutorials of MoveIt (latest version, master branch) step by step, everything is ok until I run pick_place_tutorial. The program fails to run, exits in the beginning.
> 
> ### Environment
> * ROS Distro: Kinetic
> * OS Version: Ubuntu 16.04
> * Source build (following [Source Install](https://moveit.ros.org/install/source/))
> * tag: d8678597530a8cbdb4af5075cd230fdaf1af6aae
> 
> ### Steps to reproduce
> I went through the laster version [toturial](https://ros-planning.github.io/moveit_tutorials/index.html) step by step, everything is ok until Pick and Place toturial. Both melodic-devel and kinetic-devel of [panda_moveit_config](https://github.com/ros-planning/panda_moveit_config) have been tested, and I got the same result.
> 
> ### Console output
> [Console output](https://gist.github.com/JS00000/84bd2ea37549956a866ee52d29a7612c)

Same problem.

VitaminLL, 2020-05-20:
I follow the Download Source Code in https://moveit.ros.org/install/source/. And re- catkin build moveit!. Now I can run this tutorial. I don't know why.

rhaschke, 2020-05-20:
We have seen this error (`eigen: too many iterations in Jacobi transform`) in the context of wrong input quaternions recently. A recent commit (https://github.com/ros-planning/moveit/pull/2089) provided a fix for this, which might be the reason that it works if you build MoveIt master from source.
Do you see a warning message on the console regarding empty quaternions?

felixvd, 2021-04-05:
Sounds like this is solved.

closed 2021-04-05 by felixvd