moveit/moveit_tutorials #441: Documentation of the setup assistant is not consistent on adding joints or links to the hand(Step 4)

keszegrobert, 2020-01-02:
### Description
Here:
http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html
the image:
http://docs.ros.org/melodic/api/moveit_tutorials/html/_images/setup_assistant_panda_planning_groups_gripper.png
is not in sync with its text.
In the text above the image there are Links added to the group,
in the image there are joints added.

The text does not mention if the Kinematic solver should remain None or something else.
This should confuse the readers who are not familiar with the setup assistant yet(like me)
Kinetic and Melodic are also affected.

### Your environment
* ROS Distro: [Melodic|Kinetic]
* OS Version: e.g. Ubuntu 18.04
* 1843c36 on 12 Jun 2019

### Steps to reproduce
1. Go to
http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html#step-4-add-planning-groups
2. read the text after "Add the gripper"
3. look at the picture after the text

### Expected behaviour
The image should be in sync with the text above

### Backtrace or Console output
None

rhaschke, 2020-01-04:
Thanks for pointing this out. I filed a PR to fix this. You are cordially invited to redo all the other screenshots as well as you follow the tutorial to improve their readability.

felixvd, 2021-02-10:
Fixed by #443

closed 2021-02-10 by felixvd