moveit/moveit_tutorials #444: Does not build on WIN10

CesMak, 2020-01-07:
Hey there,
trying to build moveit_tutorials on win10 but it fails:

**catkin_make --only-pkg-with-deps moveit_tutorials**

Any ideas how to solve this issue?

```

[  6%] Built target _setup_util.py_executable_install_python
[ 13%] Built target ros_api_tutorial
[ 20%] Built target robot_model_and_robot_state_tutorial
[ 26%] Built target planning_scene_tutorial
[ 33%] Built target planning_scene_ros_api_tutorial
[ 40%] Built target motion_planning_api_tutorial
[ 46%] Built target motion_planning_pipeline_tutorial
[ 60%] Built target interactivity_utils
[ 66%] Built target visualizing_collisions_tutorial
[ 73%] Built target move_group_interface_tutorial
[ 80%] Built target state_display_tutorial
[ 86%] Built target pick_place_tutorial
[ 90%] Building CXX object moveit_tutorials/doc/perception_pipeline/CMakeFiles/cylinder_segment.dir/src/cylinder_segment.cpp.obj
cylinder_segment.cpp
C:\opt\rosdeps\x64\include\boost-1_66\boost/iostreams/positioning.hpp(96): warning C4996: 'std::fpos<_Mbstatet>::seekpos': warning STL4019: The member std::fpos::seekpos() is non-Standard, and is preserved only for compatibility with workarounds for old versions of Visual C++. It will be removed in a future release, and in this release always returns 0. Please use standards-conforming mechanisms to manipulate fpos, such as conversions to and from streamoff, or an integral type, instead. If you are receiving this message while compiling Boost.IOStreams, a fix has been submitted upstream to make Boost use standards-conforming mechanisms, as it does for other compilers. You can define _SILENCE_FPOS_SEEKPOS_DEPRECATION_WARNING to acknowledge that you have received this warning, or define _REMOVE_FPOS_SEEKPOS to remove std::fpos::seekpos entirely.
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.23.28105\include\iosfwd(73): note: Siehe Deklaration von "std::fpos<_Mbstatet>::seekpos"
C:\opt\rosdeps\x64\include\flann/util/serialization.h(484): warning C4996: 'fopen': This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.  C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\stdio.h(208): note: Siehe Deklaration von "fopen"      C:\opt\rosdeps\x64\include\flann/util/serialization.h(751): warning C4996: 'fopen': This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.  C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\stdio.h(208): note: Siehe Deklaration von "fopen"      C:\opt\rosdeps\x64\include\flann/util/saving.h(58): warning C4996: 'strcpy': This function or variable may be unsafe. Consider using strcpy_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.        C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\string.h(133): note: Siehe Deklaration von "strcpy"    C:\opt\rosdeps\x64\include\flann/util/saving.h(60): warning C4996: 'strcpy': This function or variable may be unsafe. Consider using strcpy_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.        C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\string.h(133): note: Siehe Deklaration von "strcpy"    C:\opt\rosdeps\x64\include\flann/util/logger.h(66): warning C4996: 'fopen': This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\stdio.h(208): note: Siehe Deklaration von "fopen"      C:\opt\rosdeps\x64\include\pcl-1.8\pcl/sample_consensus/model_types.h(99): warning C4996: 'pcl::SAC_SAMPLE_SIZE': This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class
C:\opt\rosdeps\x64\include\pcl-1.8\pcl/sample_consensus/model_types.h(99): note: Siehe Deklaration von "pcl::SAC_SAMPLE_SIZE"
C:\opt\ros\melodic\x64\include\visualization_msgs/Marker.h(132): error C2143: Syntaxfehler: Es fehlt "}" vor "("
C:\opt\ros\melodic\x64\include\visualization_msgs/Marker.h(140): note: Siehe Verweis auf die gerade kompilierte Klasse Vorlage-Instanziierung "visualization_msgs::Marker_<ContainerAllocator>".
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(54): note: Siehe Verweis auf die gerade kompilierte Klasse Vorlage-Instanziierung "boost::arg<9>".
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(53): note: Siehe Verweis auf die gerade kompilierte Klasse Vorlage-Instanziierung "boost::arg<8>".
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(52): note: Siehe Verweis auf die gerade kompilierte Klasse Vorlage-Instanziierung "boost::arg<7>".
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(51): note: Siehe Verweis auf die gerade kompilierte Klasse Vorlage-Instanziierung "boost::arg<6>".
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(50): note: Siehe Verweis auf die gerade kompilierte Klasse Vorlage-Instanziierung "boost::arg<5>".
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(49): note: Siehe Verweis auf die gerade kompilierte Klasse Vorlage-Instanziierung "boost::arg<4>".
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(48): note: Siehe Verweis auf die gerade kompilierte Klasse Vorlage-Instanziierung "boost::arg<3>".
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(47): note: Siehe Verweis auf die gerade kompilierte Klasse Vorlage-Instanziierung "boost::arg<2>".
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(46): note: Siehe Verweis auf die gerade kompilierte Klasse Vorlage-Instanziierung "boost::arg<1>".
C:\opt\ros\melodic\x64\include\visualization_msgs/Marker.h(132): error C2059: Syntaxfehler: "Konstante"
C:\opt\ros\melodic\x64\include\visualization_msgs/Marker.h(134): error C2143: Syntaxfehler: Es fehlt ";" vor "}"        C:\opt\ros\melodic\x64\include\visualization_msgs/Marker.h(134): error C2238: Unerwartete(s) Token vor ";"
NMAKE : fatal error U1077: "C:\PROGRA~2\MIB055~1\2019\COMMUN~1\VC\Tools\MSVC\1423~1.281\bin\Hostx64\x64\cl.exe": Rückgabe-Code "0x2"
Stop.
NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.23.28105\bin\HostX64\x64\nmake.exe"": Rückgabe-Code "0x2"
Stop.
NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.23.28105\bin\HostX64\x64\nmake.exe"": Rückgabe-Code "0x2"
Stop.
Invoking "nmake" failed
```

rhaschke, 2020-01-08:
Looks like the generated message header `include\visualization_msgs/Marker.h` is broken.
Either the generation is indeed broken (which is unlikely) or there was a build race condition, because package dependencies were not correctly modeled.
Could you try to build `visualization_msgs` first? If the failure remains, the generation is broken and you should report the issue at https://github.com/ros/gencpp.

CesMak, 2020-01-09:
Hm well under my Ubuntu 16.04 (kinetic) all works fine. 

Well I think I already have on Win10 (melodic)
`
C:\Windows\System32>roscd visualization_msgs                                                                                                                                                                                                    C:\opt\ros\melodic\x64\share\visualization_msgs>    `

Or do I need an other visualization msgs package?

rhaschke, 2020-01-09:
No, that's the only `visualization_msgs` package available and needed.
However, the error message clearly states:
```
C:\opt\ros\melodic\x64\include\visualization_msgs/Marker.h(132): error C2143: Syntaxfehler: Es fehlt "}" vor "("
C:\opt\ros\melodic\x64\include\visualization_msgs/Marker.h(140): note: Siehe Verweis auf die gerade kompilierte Klasse Vorlage-Instanziierung "visualization_msgs::Marker_<ContainerAllocator>".
...
C:\opt\ros\melodic\x64\include\visualization_msgs/Marker.h(132): error C2059: Syntaxfehler: "Konstante"
C:\opt\ros\melodic\x64\include\visualization_msgs/Marker.h(134): error C2143: Syntaxfehler: Es fehlt ";" vor "}"        C:\opt\ros\melodic\x64\include\visualization_msgs/Marker.h(134): error C2238: Unerwartete(s) Token vor ";"
```
which indicates that the issue is in those headers. Could you post those lines? 
Try to reinstall the `visualization_msgs` package?

CesMak, 2020-01-09:
Hm I also thought that I need to install:

C:\Windows\System32>**choco upgrade ros-melodic-moveit-visual-tools**                                                       `Chocolatey v0.10.15                                                                                                     Upgrading the following packages:                                                                                       ros-melodic-moveit-visual-tools                                                                                         By upgrading you accept licenses for the packages.                                                                      ros-melodic-moveit-visual-tools is not installed. Installing...                                                         ros-melodic-moveit-visual-tools not installed. The package was not found with the source(s) listed.   `

Cause I required this package when using kinetic in ubuntu. But it also fails to install.

rhaschke, 2020-01-09:
[`moveit-visual-tools`](https://github.com/ros-planning/moveit_visual_tools) is a utility package, e.g. used in `moveit_tutorials`. It's not required by MoveIt itself.

CesMak, 2020-02-18:
Did anyone of you find a solution ? I still found none.....

lilustga, 2020-02-26:
C:\opt\ros\melodic\x64\include\visualization_msgs/Marker.h(132): error C2059: syntax error: 'constant'

I looked into the above error and the issue originates in C:\ws_moveit\src\moveit_tutorials\doc\perception_pipeline\src\cylinder_segment.cpp

A file which included the window.h header (pcl) was included before a file which included (marker.h via planning_scene_interface.h). This is problematic because marker.h defines DELETE as an enum member after DELETE was defined as a macro in windows.h.

PR with a fix coming shortly