moveit/moveit_tutorials #453: Running OMPL as a pre-processor for STOMP dosen't work.

luckychay, 2020-02-17:
### Description

I'm following the Planning Adapter Tutorials of Moveit! at http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/planning_adapters/planning_adapters_tutorial.html, but I can't implement the Running OMPL as a pre-processor for STOMP. Although I installed stomp_core from source which origined from https://github.com/ros-industrial/stomp_ros, I got an error that `stomp/SmoothingAdapter` doesn't exist.

### Your environment
* ROS Distro: [Kinetic]
* OS Version:  Ubuntu 16.04 LTS

### Steps to reproduce
just follow the tutorials and run :
`roslaunch panda_moveit_config demo.launch`

### Expected behaviour
The OMPL and STOMP works together to get a solution of the manipulator without error.

###  Console output

`[ERROR] [1581905685.671780997]: Exception while loading planning adapter plugin 'stomp/SmoothingAdapter': According to the loaded plugin descriptions the class stomp/SmoothingAdapter with base class type planning_request_adapter::PlanningRequestAdapter does not exist. Declared types are chomp/OptimizerAdapter default_planner_request_adapters/AddIterativeSplineParameterization default_planner_request_adapters/AddTimeOptimalParameterization default_planner_request_adapters/AddTimeParameterization default_planner_request_adapters/Empty default_planner_request_adapters/FixStartStateBounds default_planner_request_adapters/FixStartStateCollision default_planner_request_adapters/FixStartStatePathConstraints default_planner_request_adapters/FixWorkspaceBounds default_planner_request_adapters/ResolveConstraintFrames industrial_trajectory_filters/AddSmoothingFilter industrial_trajectory_filters/NPointFilter industrial_trajectory_filters/UniformSampleFilter`

rhaschke, 2020-02-17:
The error message states that `stomp/SmoothingAdapter` is not found by the ROS plugin mechanism.
Verify that the corresponding plugin description XML is correctly installed and found by ROS.

luckychay, 2020-02-17:
@rhaschke Thank you for your reply. I do think  `stomp/SmoothingAdapter` is not found  due to  the uncorrect installation of the plugin description XML. The point is that I can not find anything related to  `stomp/SmoothingAdapter` within the respority I pulled from [ https://github.com/ros-industrial/stomp_ros](url) or the one at [https://github.com/ros-industrial/industrial_moveit](url). I'm wondering if it has not been well developed? Then why not remove this part from the tutorials?

PR #488 (2020-05-11): Fix name of STOMP smoothing adapter
STOMP smoothing adapter was merged into stomp_ros today: https://github.com/ros-industrial/stomp_ros/pull/20

This fixes the name of the exported plugin and therefore fixes #453
closed 2020-05-11 by v4hn