moveit/moveit_tutorials #469: moveit_tutorials error on catkin build command

femifapo, 2020-03-08:
### Description
I am trying to install the moveit tutorial but i fails on the "catkin build" command.
 

Overview of your issue here.
use the "catkin build" command
### Your environment
* ROS Distro: [melodic]
* OS Version: e.g. Ubuntu 18

### Steps to reproduce
Just following steps here -> http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/getting_started/getting_started.html

and it fails on here ->Build your Catkin Workspace  and fails on this command catkin build

### Expected behaviour
Tell us what should happen

### Backtrace or Console output

femi@femi-VirtualBox:~/ws_moveit$ catkin config --extend /opt/ros/${ROS_DISTRO} --cmake-args -DCMAKE_BUILD_TYPE=Release
---------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/melodic
Workspace:                   /home/femi/ws_moveit
---------------------------------------------------------
Build Space:        [exists] /home/femi/ws_moveit/build
Devel Space:        [exists] /home/femi/ws_moveit/devel
Install Space:      [unused] /home/femi/ws_moveit/install
Log Space:          [exists] /home/femi/ws_moveit/logs
Source Space:       [exists] /home/femi/ws_moveit/src
DESTDIR:            [unused] None
---------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------
femi@femi-VirtualBox:~/ws_moveit$ catkin build
---------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/melodic
Workspace:                   /home/femi/ws_moveit
---------------------------------------------------------
Build Space:        [exists] /home/femi/ws_moveit/build
Devel Space:        [exists] /home/femi/ws_moveit/devel
Install Space:      [unused] /home/femi/ws_moveit/install
Log Space:          [exists] /home/femi/ws_moveit/logs
Source Space:       [exists] /home/femi/ws_moveit/src
DESTDIR:            [unused] None
---------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------
[build] Found '2' packages in 0.0 seconds.                                                                               
[build] Package table is up to date.                                                                                     
Starting  >>> panda_moveit_config                                                                                        
Finished  <<< panda_moveit_config                [ 0.1 seconds ]                                                         
Starting  >>> moveit_tutorials                                                                                           
_________________________________________________________________________________________________________________________
Errors     << moveit_tutorials:make /home/femi/ws_moveit/logs/moveit_tutorials/build.make.001.log                        
/home/femi/ws_moveit/src/moveit_tutorials/doc/interactivity/src/interactive_robot.cpp: In member function ‘void InteractiveRobot::updateAll()’:
/home/femi/ws_moveit/src/moveit_tutorials/doc/interactivity/src/interactive_robot.cpp:235:76: warning: ‘bool moveit::core::RobotState::setFromIK(const moveit::core::JointModelGroup*, const Isometry3d&, unsigned int, double, const GroupStateValidityCallbackFn&, const kinematics::KinematicsQueryOptions&)’ is deprecated: The attempts argument is not supported anymore. [-Wdeprecated-declarations]
   if (robot_state_->setFromIK(group_, desired_group_end_link_pose_, 10, 0.1))
                                                                            ^
In file included from /home/femi/ws_moveit/src/moveit_tutorials/doc/interactivity/include/interactivity/interactive_robot.h:45:0,
                 from /home/femi/ws_moveit/src/moveit_tutorials/doc/interactivity/src/interactive_robot.cpp:39:
/opt/ros/melodic/include/moveit/robot_state/robot_state.h:995:3: note: declared here
   setFromIK(const JointModelGroup* group, const Eigen::Isometry3d& pose, unsigned int attempts, double timeout = 0.0,
   ^~~~~~~~~
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [doc/motion_planning_api/CMakeFiles/motion_planning_api_tutorial.dir/src/motion_planning_api_tutorial.cpp.o] Error 4
make[1]: *** [doc/motion_planning_api/CMakeFiles/motion_planning_api_tutorial.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/femi/ws_moveit/build/moveit_tutorials; catkin build --get-env moveit_tutorials | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.........................................................................................................................
Failed     << moveit_tutorials:make              [ Exited with code 2 ]                                                  
Failed    <<< moveit_tutorials                   [ 16 minutes and 11.3 seconds ]                                         
[build] Summary: 1 of 2 packages succeeded.                                                                              
[build]   Ignored:   None.                                                                                               
[build]   Warnings:  None.                                                                                               
[build]   Abandoned: None.                                                                                               
[build]   Failed:    1 packages failed.                                                                                  
[build] Runtime: 16 minutes and 11.5 seconds total.

rhaschke, 2020-03-08:
Looks like, your machine is running out of memory when compiling MoveIt. Try limiting the number of parallel jobs:
`catkin build --jobs 1`

femifapo, 2020-03-12:
You were right. I increased the allocated memory and it completed just fine. Thank you.

closed 2020-03-12 by femifapo

stefanieZhao77, 2021-01-27:
> Looks like, your machine is running out of memory when compiling MoveIt. Try limiting the number of parallel jobs:
> `catkin build --jobs 1`

It works, thank you!