moveit/moveit_tutorials #470: Motion planning pipeline tutorial roslaunch Error

edetleon, 2020-03-11:
### Description

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04 on VirtualBox
* Binary build
* Unversionned

### Steps to reproduce

- Install moveit_tutorials package with git clone in /src :
`git clone https://github.com/ros-planning/moveit_tutorials.git -b melodic-devel`

- Make your environnement (catkin_make), and follow this tutorial : http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/motion_planning_pipeline/motion_planning_pipeline_tutorial.html

- At the first step, roslaunch the launch file. A joint_state_publisher GUI appears, and safter a few seconds you can see the panda robot in the Rviz GUI :
`roslaunch moveit_tutorials motion_planning_pipeline_tutorial.launch`

- When the terminal prints `Waiting to continue: Press 'next' in the RvizVisualToolsGui window to start the demo... continuing` press the Next button in RvizVisualToolsGui (bottom left of the Rviz window).

### Expected behaviour

The panda arm moves to a pose goal.

### Actual behaviour

In the Rviz GUI, the arm is in collision state.
Error messages are displayed on the terminal ant the panda arm doesn't move.

I don't know how the joint_state_publisher GUI is supposed to work (I never used it), but when I change the joints values the robot doesn't move.

### Backtrace or Console output

```
WARNING: Package name "swurdf_Arm_ikfast_plugin" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "patte_ikfast_Leg_plugin" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
process[virtual_joint_broadcaster_0-1]: started with pid [14502]
process[joint_state_publisher-2]: started with pid [14503]
process[robot_state_publisher-3]: started with pid [14504]
process[rviz_edith_VirtualBox_14481_8276606921159861290-4]: started with pid [14505]
process[motion_planning_pipeline_tutorial-5]: started with pid [14511]
[ INFO] [1583937715.047109946]: Loading robot model 'panda'...
[ INFO] [1583937715.150063252]: rviz version 1.13.7
[ INFO] [1583937715.150168630]: compiled against Qt version 5.9.5
[ INFO] [1583937715.150211600]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1583937715.170898961]: Forcing OpenGl version 0.
[ WARN] [1583937715.240433455]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/panda_arm/kinematics_solver_attempts' from your configuration.
[ INFO] [1583937715.403445120]: Initializing OMPL interface using ROS parameters
[ INFO] [1583937715.432621290]: Using planning interface 'OMPL'
[ INFO] [1583937715.435969169]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1583937715.437572740]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1583937715.438574314]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1583937715.439092635]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1583937715.439548150]: Param 'jiggle_fraction' was not set. Using default value: 0.02
[ INFO] [1583937715.443451599]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1583937715.443654107]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1583937715.443784017]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1583937715.443904599]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1583937715.444013229]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1583937715.444125606]: Using planning request adapter 'Fix Start State Path Constraints'
[ INFO] [1583937715.451292139]: RemoteControl Ready.
[ INFO] [1583937715.609301891]: Stereo is NOT SUPPORTED
[ INFO] [1583937715.609644307]: OpenGl version: 3.1 (GLSL 1.4).
[ERROR] [1583937715.772605028]: Ignoring transform for child_frame_id "100" from authority "unknown_publisher" because of an invalid quaternion in the transform (0,000000 0,000000 0,000000 0,000000)
[ WARN] [1583937715.952220903]: Topic '/rviz_visual_tools' unable to connect to any subscribers within 0.5 sec. It is possible initially published visual messages will be lost.
[ INFO] [1583937715.958582894]: Loading robot model 'panda'...
[WARN] [1583937716.073134]: The 'use_gui' parameter was specified, which is deprecated.  We'll attempt to find and run the GUI, but if this fails you should install the 'joint_state_publisher_gui' package instead and run that.  This backwards compatibility option will be removed in Noetic.
[ INFO] [1583937716.888517373]: Loading robot model 'panda'...
[ WARN] [1583937716.966021391]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/rviz_edith_VirtualBox_14481_8276606921159861290/panda_arm/kinematics_solver_attempts' from your configuration.
[ INFO] [1583937717.073596753]: Starting planning scene monitor
[ INFO] [1583937717.076104397]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1583937717.077392510]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ INFO] [1583937722.097813835]: Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-melodic-moveit-ros-planning-1.0.2/planning_scene_monitor/src/planning_scene_monitor.cpp:495

Waiting to continue: Press 'next' in the RvizVisualToolsGui window to start the demo... continuing
[ERROR] [1583937768.544020072]: Found empty JointState message
[ERROR] [1583937768.544548844]: Found empty JointState message
[ INFO] [1583937768.544964896]: Found a contact between 'panda_hand' (type 'Robot link') and 'panda_link5' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1583937768.545126177]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1583937768.545499863]: Start state appears to be in collision with respect to group panda_arm
[ WARN] [1583937768.610460199]: Unable to find a valid state nearby the start state (using jiggle fraction of 0.020000 and 100 sampling attempts). Passing the original planning request to the planner.
[ERROR] [1583937768.614835703]: Found empty JointState message
[ERROR] [1583937768.615512649]: Found empty JointState message
[ INFO] [1583937768.617181032]: The timeout for planning must be positive (0.000000 specified). Assuming one second instead.
[ INFO] [1583937768.617841086]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ WARN] [1583937768.618421402]: RRTConnect: Skipping invalid start state (invalid state)
[ERROR] [1583937768.618601936]: RRTConnect: Motion planning start tree could not be initialized!
[ INFO] [1583937768.618724206]: No solution found after 0.000380 seconds
[ WARN] [1583937768.620438744]: Goal sampling thread never did any work.
[ INFO] [1583937768.620644016]: Unable to solve the planning problem
[ERROR] [1583937768.621580727]: Could not compute plan successfully
[motion_planning_pipeline_tutorial-5] process has finished cleanly
log file: /home/edith/.ros/log/09305068-639b-11ea-90ad-0800273c0207/motion_planning_pipeline_tutorial-5*.log

```

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

henningkayser, 2020-06-04:
@edetleon thanks for this report. Did you clone the panda_moveit_config package into your workspace as well as described in the Getting Started section? This issue has been fixed a while ago https://github.com/ros-planning/panda_moveit_config/pull/47. I assume the setup was picking up an outdated version of that package.

simonbogh, 2021-01-21:
I can confirm this is still happening with the same setup. `panda_moveit_config` is cloned into the workspace.

Getting this error as well, which is probably due to the former

    ros.moveit_ros_visualization: Unable to connect to move_group action server 'move_group' within allotted time (30s)

Baschalp, 2021-02-04:
I could also confirm the issue having the panda_moveit_config being cloned into the ws. Solution mentioned [here](https://github.com/ros-planning/moveit_tutorials/pull/273#issue-240611365) did not work on Ubuntu 18.04 (VirtualBox) with ROS Melodic.

OmkarKabadagi5823, 2021-05-17:
Even I faced the same issue in the tutorial, but I was able to fix it with few modifications in code and launch file.
Fix for the first issue was to load the arm in a valid state. That can be done by setting the state to `ready` using planning scene in code. Planning the pose now works after this edit. But this still does visually change the state of arm in RViz, but if you plan a pose with `show trail` enabled for trajectory, you can see the trail starts from `ready` state instead of the collision state displayed which means the change in state is not being updated visually.
After seeing the `rqt_graph`, I realised that the `/joint_states` is not being taken as a feedback any where except `robot_state_publisher`. Including `move_group.launch` and remapping `/joint_states` to `/joints_states_desired` (move_group subscribes to `/joint_state_desired` to distinguish input and feedback) fixes the second issue.