moveit/moveit_tutorials #471: How can I get the moveit_msgs/Grasp.msg when picking with planning only?

CaffeMrDe, 2020-03-14:
### Description

I test pick_place_tutorial in the tutorials project.  I set the pick funtion arg Planonly  as true ,it is sucessfully . I try to store this value which was got to planning path and I can set the it for target to excuate the robot motion .

Because I want  to the  excuate  robot motion preferly  everytime. I try to get helpful information on the tutorial net , but it unuse.

Can you help me please? 
Thank you!

### Your environment
* ROS Distro: [Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build?

DHeroPlasma, 2020-04-05:
I had a similar problem after finishing the MotionPlanningAPI tutorials and trying to apply the code to my case. Maybe this will help you:

planning_interface::MotionPlanRequest req
planning_interface::MotionPlanResponse res;
moveit_msgs::MotionPlanResponse response;

planning_interface::PlanningContextPtr context = planner_instance->getPlanningContext(planning_scene, req, res.error_code_);

res.getMessage(response);
moveit::planning_interface::MoveGroupInterface::Plan my_plan;
my_plan.trajectory_ = response.trajectory;
move_group->execute(my_plan);

Greetings

CaffeMrDe, 2020-04-09:
However , I want to get the trajectory with applying the following code:

`
std::vector<moveit_msgs::Grasp> grasps;
grasps.resize(1);

/* Add both finger joints of panda robot. */
posture.joint_names.resize(2);
posture.joint_names[0] = "panda_finger_joint1";
posture.joint_names[1] = "panda_finger_joint2";

/* Set them as closed. */
posture.points.resize(1);
posture.points[0].positions.resize(2);
posture.points[0].positions[0] = 0.00;
posture.points[0].positions[1] = 0.00;
posture.points[0].time_from_start = ros::Duration(0.5);

move_group.pick("object", grasps);   // get the trajectory 
`

Are you thinking about how to solve the problem?
 Thanks~

felixvd, 2021-04-05:
Closing for inactivity. Feel free to open if the problem persists.

closed 2021-04-05 by felixvd