moveit/moveit_tutorials #475: Environment issue

JohnsonQi, 2020-03-23:
### Description

Overview of your issue here.
I want to simulate.
However, I can't successfully run : roslaunch panda_autograsp panda_autograsp.py

### Your environment
* ROS Distro: [Indigo|Jade|Kinetic] Kinetic
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

After 
catkin build -j4 -DCMAKE_BUILD_TYPE=Release -DFranka_DIR:PATH=/libfranka/build -Dfreenect2_DIR=/opt/freenect2/lib/cmake/freenect2"
cd ~/panda_autograsp_ws/src
pip install .

run：roslaunch panda_autograsp panda_autograsp.py

Part1   (I guess it's caused by no real kinect v2)
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 GLX: Failed to create context: GLXBadFBConfig
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.
[FATAL] [1584971719.851185887]: Failed to load nodelet '/kinect2_points_xyzrgb_hd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'
[FATAL] [1584971719.851311419]: Failed to load nodelet '/kinect2_depth_image_proc_hd` of type `depth_image_proc/convert_metric` to manager `kinect2'
[FATAL] [1584971719.851379355]: Failed to load nodelet '/kinect2_depth_image_proc_sd` of type `depth_image_proc/convert_metric` to manager `kinect2'
[FATAL] [1584971719.851382338]: Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'
[FATAL] [1584971719.851392479]: Failed to load nodelet '/kinect2_depth_image_proc_qhd` of type `depth_image_proc/convert_metric` to manager `kinect2'
[FATAL] [1584971719.851464927]: Failed to load nodelet '/kinect2_points_xyzrgb_sd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'
[FATAL] [1584971719.851546718]: Failed to load nodelet '/kinect2_points_xyzrgb_qhd` of type `depth_image_proc/point_cloud_xyzrgb` to manager `kinect2'

Part2 (Don't know what happened)
[ INFO] [1584971719.892685049]: compiled against OGRE version 1.9.0 (Ghadamon)
Traceback (most recent call last):
  File "/home/johnson/panda_autograsp_ws/src/panda_autograsp/nodes/tf2_broadcaster.py", line 21, in <module>
    from panda_autograsp import Tf2Broadcaster
  File "/home/johnson/panda_autograsp_ws/devel/lib/python2.7/dist-packages/panda_autograsp/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 2, in <module>
  File "/home/johnson/panda_autograsp_ws/src/panda_autograsp/src/panda_autograsp/loggers.py", line 22, in <module>
    import colorlog
ImportError: No module named colorlog
Traceback (most recent call last):
  File "/home/johnson/panda_autograsp_ws/src/panda_autograsp/nodes/moveit_planner_server.py", line 44, in <module>
    from panda_autograsp import MoveitPlannerServer
  File "/home/johnson/panda_autograsp_ws/devel/lib/python2.7/dist-packages/panda_autograsp/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 2, in <module>
  File "/home/johnson/panda_autograsp_ws/src/panda_autograsp/src/panda_autograsp/loggers.py", line 22, in <module>
    import colorlog
ImportError: No module named colorlog
Traceback (most recent call last):
  File "/home/johnson/panda_autograsp_ws/src/panda_autograsp/nodes/panda_autograsp_server.py", line 32, in <module>
    from panda_autograsp import PandaAutograspServer
  File "/home/johnson/panda_autograsp_ws/devel/lib/python2.7/dist-packages/panda_autograsp/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 2, in <module>
  File "/home/johnson/panda_autograsp_ws/src/panda_autograsp/src/panda_autograsp/loggers.py", line 22, in <module>
    import colorlog
ImportError: No module named colorlog

Part3：================================================================================REQUIRED process [tf2_broadcaster-16] has died!
process has died [pid 39153, exit code 1, cmd /home/johnson/panda_autograsp_ws/src/panda_autograsp/nodes/tf2_broadcaster.py panda_link0 calib_frame __name:=tf2_broadcaster __log:=/home/johnson/.ros/log/e054c018-6d02-11ea-9c9f-00505621cb8c/tf2_broadcaster-16.log].
log file: /home/johnson/.ros/log/e054c018-6d02-11ea-9c9f-00505621cb8c/tf2_broadcaster-16*.log
Initiating shutdown!
================================================================================

Part4:[moveit_planner_server-19] killing on exit
[grasp_planner_server-18] killing on exit
Traceback (most recent call last):
[... 53 lines cut ...]
  File "/home/johnson/envs/ros-grasp2/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2766, in _get_metadata
    if self.has_metadata(name):
  File "/home/johnson/envs/ros-grasp2/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1891, in has_metadata
[joint_state_desired_publisher-3] killing on exit
    return name == 'PKG-INFO' and os.path.isfile(self.path)
  File "/usr/lib/python2.7/genericpath.py", line 37, in isfile
[kinect2_points_xyzrgb_hd-11] killing on exit
[kinect2_points_xyzrgb_sd-9] killing on exit
[virtual_joint_broadcaster_1-1] killing on exit
[kinect2_bridge-8] killing on exit
[kinect2-7] killing on exit
[joint_state_publisher-2] killing on exit
    st = os.stat(path)

part5:
PluginManager._discover() force discovery of plugins
RospkgPluginProvider._find_plugins() crawling for plugins of type 'qt_gui'
[rviz_johnson_VM_UBUNTU16_38979_7231073059793820187-5] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done

Could you please tell me what I should do next?

Looking forward to your reply

JohnsonQi, 2020-03-23:
python version 2.7

v4hn, 2020-03-23:
> However, I can't successfully run : roslaunch panda_autograsp panda_autograsp.py

What? Who told you to run this? No references to any tutorials sounds like this issue does not belong here.

JohnsonQi, 2020-03-23:
sorry, sir. I got the wrong page

closed 2020-03-23 by v4hn