moveit/moveit_tutorials #478: Improvements to ompl benchmarking tutorial (melodic)

simonschmeisser, 2020-04-07:
### Description

I tried following the benchmarking tutorial and ran into some issues, collecting them here first and hope to address them in PRs

- [ ] mongodb not actually installed: https://github.com/ros-planning/warehouse_ros_mongo/issues/31
- [ ] tutorial claims it needs from source build but everything is released
- [ ] tutorial names query `Pick1` while default benchmark file calls it `KitchenPick1`
- [ ] clarify in tutorial that query should be for group `panda_arm`
- [ ] default projection missing? 

```
[ INFO] [1586249659.474563951]: Planner configuration 'panda_arm[BKPIECEkConfigDefault]' will use planner 'geometric::BKPIECE'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1586249659.474780724]: panda_arm[BKPIECEkConfigDefault]: Attempting to use default projection.
[ERROR] [1586249659.474879369]: No default projection is set. Perhaps setup() needs to be called
[ERROR] [1586249659.478963309]: OMPL encountered an error: No projection evaluator specified
================================================================================REQUIRED process [moveit_run_benchmark-3] has died!

```
- [ ] use sqlite instead of mongodb? https://github.com/k-okada/warehouse_ros_sqlite

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source
* If source, which git commit or tag? current

simonschmeisser, 2020-04-07:
default projection can be fixed by adding:

projection_evaluator: joints(panda_joint1,panda_joint2) to ompl_planning.yaml

simonschmeisser, 2020-05-20:
@rhaschke could yout tag this as WorldMoveItDay candidate so whoever is interested can find it?

mamoll, 2020-06-16:
The benchmarking functionality is actually not OMPL-specific. Other planning pipelines (CHOMP, STOMP, ...) can also be used. This tutorial has lots of issues as @simonschmeisser has already pointed out. I took me a while to get things working, too.

felixvd, 2021-03-14:
In addition to the thread @simonschmeisser just linked from, I noticed that the `Stored Scenes` and `Stored States` tabs are never mentioned or used outside of the Benchmarking tutorial page. The only use "in the wild" I have seen is in [this PR](https://github.com/ros-planning/moveit/pull/932), for which there are good reasons not to merge imo (although the functionality has a use). 

I would consider what the intended workflow  for setting up a benchmarking problem is, and then try to combine the Stored Scenes/Stored States tab into one if possible. I actually also think the `warehouse` connection on the connect tab should be in there, because it affects nothing except those two tabs.

Note that the scripts in [this package](https://planners-benchmarking.readthedocs.io/en/latest/user_guide/4_running_an_experiment.html) allow saving scenes/states to files, which is more intuitive and much easier to share/add to a repo.