moveit/moveit_tutorials #483: Building MoveIt Dependencies: sudo scons install

DanielSchimpf, 2020-04-23:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Kinetic]
* OS Version:  Ubuntu 16.04
* Binary build

I have been following the instructions to build the moveit dependencies, based on the instructions found here: https://moveit.ros.org/install/source/dependencies/

Everything works fine until: sudo scons install

Upon that I get:
```
dan@dan-VirtualBox:~/mongo-cxx-driver$ sudo scons install
[sudo] password for dan: 
scons: Reading SConscript files ...
scons version: 2.4.1
python version: 2 7 12 'final' 0
Checking whether the C++ compiler works... yes
Checking whether the C compiler works... yes
Checking if C++ compiler "g++" is GCC... yes
Checking if C compiler "gcc" is GCC... yes
Checking if C compiler supports -Wno-unused-local-typedefs... yes
Checking if C compiler supports -Wno-unused-function... yes
Checking if C compiler supports -Wno-unused-private-field... no
Checking if C compiler supports -Wno-deprecated-declarations... yes
Checking if C compiler supports -Wno-tautological-constant-out-of-range-compare... no
Checking if C compiler supports -Wno-unused-const-variable... no
Checking if C compiler supports -fno-builtin-memcmp... yes
Checking for __thread... yes
Checking for C header file unistd.h... yes
Checking whether clock_gettime is declared... yes
Checking for C library rt... yes
Checking for C++ header file execinfo.h... yes
Checking whether backtrace is declared... yes
Checking whether backtrace_symbols is declared... yes
Checking whether backtrace_symbols_fd is declared... yes
Checking for C library pcap... no
Checking for C library wpcap... no
Checking whether timegm is declared... yes
Error: You must use --use-system-boost when building the C++ driver
```

I tried to fix that by adding  "--use-system-boost", but that creates a new error:
```
dan@dan-VirtualBox:~/mongo-cxx-driver$ sudo scons install --use-system-boost
scons: Reading SConscript files ...
scons version: 2.4.1
python version: 2 7 12 'final' 0
Checking whether the C++ compiler works... yes
Checking whether the C compiler works... yes
Checking if C++ compiler "g++" is GCC... yes
Checking if C compiler "gcc" is GCC... yes
Checking if C compiler supports -Wno-unused-local-typedefs... yes
Checking if C compiler supports -Wno-unused-function... yes
Checking if C compiler supports -Wno-unused-private-field... no
Checking if C compiler supports -Wno-deprecated-declarations... yes
Checking if C compiler supports -Wno-tautological-constant-out-of-range-compare... no
Checking if C compiler supports -Wno-unused-const-variable... no
Checking if C compiler supports -fno-builtin-memcmp... yes
Checking for __thread... yes
Checking for C++ header file boost/filesystem/operations.hpp... yes
Checking for C++ library boost_thread-mt... no
Checking for C++ library boost_thread... yes
Checking for C++ library boost_filesystem-mt... no
Checking for C++ library boost_filesystem... yes
Checking for C++ library boost_program_options-mt... no
Checking for C++ library boost_program_options... yes
Checking for C++ library boost_system-mt... no
Checking for C++ library boost_system... yes
Checking for C header file unistd.h... yes
Checking whether clock_gettime is declared... yes
Checking for C library rt... yes
Checking for C++ header file execinfo.h... yes
Checking whether backtrace is declared... yes
Checking whether backtrace_symbols is declared... yes
Checking whether backtrace_symbols_fd is declared... yes
Checking for C library pcap... no
Checking for C library wpcap... no
Checking whether timegm is declared... yes
scons: done reading SConscript files.
scons: Building targets ...
scons: *** Do not know how to make File target `install' (/home/dan/mongo-cxx-driver/install).  Stop.
scons: building terminated because of errors.
```

The reason why I want to install the MongoDB Warehouse Support is because I want to follow the benchmarking tutorial ([](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/benchmarking/benchmarking_tutorial.html)), which links to the dependency page.

Do you guys know what to do? Let me know if i should add any other information.

Thanks!

rhaschke, 2020-04-24:
Probably the documentation how to build the mongo driver from source is outdated.
Can you please link the original tutorial page which was referring to the source install?

At least in Melodic, everything works fine using the binary release of `libmongoclient-dev`. Did you try that first? If that doesn't work, please check recent build instructions for Mongo on their site.
If anything changed there, please report back and (optimally) file a PR for https://github.com/ros-planning/moveit.ros.org to update our documentation.

DanielSchimpf, 2020-04-24:
Thanks for your quick response!
I did not try the binary release yet, but I will now.  And the tutorial which refers to the source install is the benchmarking tutorial (http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/benchmarking/benchmarking_tutorial.html): 

> To use this benchmarking method, you will need to download and install the ROS Warehouse plugin. Currently this is not available from Debians and requires a source install for at least some aspects. For source instructions, see this page

mlautman, 2020-05-05:
@DanielSchimpf Please let us know if this fixes your issue so we can close this