moveit/moveit_tutorials #484: Benchmarking tutorial melodic error

DanielSchimpf, 2020-04-24:
### Description

Errors while following the benchmarking tutorial (https://ros-planning.github.io/moveit_tutorials/doc/benchmarking/benchmarking_tutorial.html).

### Your environment
* ROS Distro: Melodic
* OS Version:  Ubuntu 18
* Source  build, based on this https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html

### Steps to reproduce
Install from source based on this tutorial: https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html
Install MongoDB: `sudo apt-get install libmongoclient-dev`
Adding warehouse_ros and warehouse_ros_mongo packages and build: (https://moveit.ros.org/install/source/dependencies/) [This seems to be a dependency of the tutorial, is mentioned on the kinetic tutorial website: http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/benchmarking/benchmarking_tutorial.html]
Running first command from tutorial (https://ros-planning.github.io/moveit_tutorials/doc/benchmarking/benchmarking_tutorial.html) `roslaunch panda_moveit_config demo.launch db:=true`, adding MotionPlanning and clicking connect causes a `std::bad_alloc` error. 

### Backtrace or Console output

<details>
  <summary>run log</summary>

```
dan@dan-VirtualBox:~/ws_moveit/src$ roslaunch panda_moveit_config demo.launch db:=true
... logging to /home/dan/.ros/log/a58f850e-863d-11ea-80d2-080027fb9338/roslaunch-dan-VirtualBox-29764.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dan-VirtualBox:41213/

SUMMARY
========

PARAMETERS
 * /joint_state_publisher/use_gui: False
 * /mongo_wrapper_ros_dan_VirtualBox_29764_4383733543191741152/database_path: /home/dan/ws_move...
 * /mongo_wrapper_ros_dan_VirtualBox_29764_4383733543191741152/overwrite: False
 * /move_group/allow_trajectory_execution: True
 * /move_group/controller_list: [{'joints': ['pan...
 * /move_group/hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/initial: [{'pose': 'ready'...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_fake_contr...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_frame: camera_rgb_optica...
 * /move_group/octomap_resolution: 0.05
 * /move_group/panda_arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/panda_arm_hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/planner_configs/BFMTkConfigDefault/balanced: 0
 * /move_group/planner_configs/BFMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/BFMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/BFMTkConfigDefault/heuristics: 1
 * /move_group/planner_configs/BFMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/BFMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/BFMTkConfigDefault/optimality: 1
 * /move_group/planner_configs/BFMTkConfigDefault/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMTkConfigDefault/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/BiESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiESTkConfigDefault/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRTkConfigDefault/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/init_temperature: 100
 * /move_group/planner_configs/BiTRRTkConfigDefault/range: 0.0
[... 246 lines cut ...]
[ INFO] [1587741008.127610313]: No active joints or end effectors found for group 'hand'. Make sure that kinematics.yaml is loaded in this node's namespace.
[ INFO] [1587741008.129101312]: No active joints or end effectors found for group 'hand'. Make sure that kinematics.yaml is loaded in this node's namespace.
[ INFO] [1587741008.178478656]: Constructing new MoveGroup connection for group 'hand' in namespace ''
QAbstractItemView::setSelectionModel() failed: Trying to set a selection model, which works on a different model than the view.
QAbstractItemView::setSelectionModel() failed: Trying to set a selection model, which works on a different model than the view.
[ INFO] [1587741009.405677909]: Ready to take commands for planning group hand.
[ INFO] [1587741009.405777423]: Looking around: no
[ INFO] [1587741009.405814684]: Replanning: no
QAbstractItemView::setSelectionModel() failed: Trying to set a selection model, which works on a different model than the view.
QAbstractItemView::setSelectionModel() failed: Trying to set a selection model, which works on a different model than the view.
QAbstractItemView::setSelectionModel() failed: Trying to set a selection model, which works on a different model than the view.
QAbstractItemView::setSelectionModel() failed: Trying to set a selection model, which works on a different model than the view.
QAbstractItemView::setSelectionModel() failed: Trying to set a selection model, which works on a different model than the view.
QAbstractItemView::setSelectionModel() failed: Trying to set a selection model, which works on a different model than the view.
QAbstractItemView::setSelectionModel() failed: Trying to set a selection model, which works on a different model than the view.
QAbstractItemView::setSelectionModel() failed: Trying to set a selection model, which works on a different model than the view.
QAbstractItemView::setSelectionModel() failed: Trying to set a selection model, which works on a different model than the view.
QAbstractItemView::setSelectionModel() failed: Trying to set a selection model, which works on a different model than the view.
QAbstractItemView::setSelectionModel() failed: Trying to set a selection model, which works on a different model than the view.
QAbstractItemView::setSelectionModel() failed: Trying to set a selection model, which works on a different model than the view.
[ERROR] [1587741012.223363253]: std::bad_alloc

```

</details>

mlautman, 2020-05-05:
I have had issues with this one as well. Perhaps @mamoll can give this tutorial a run-through and see if he can make clear any of the pit-falls

simonschmeisser, 2020-05-20:
see also #478 for a list of open issues