moveit/moveit_tutorials #503: Question about 'Low Level Controllers (gripper section)'

seoulstudent, 2020-06-23:
Dear Sir,

I'd like to ask you a question about moveit tutorial 'Low Level Controllers (gripper section)'
(http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/controller_configuration/controller_configuration_tutorial.html)

I wrote as follows 'controllers.yaml' file in 'moveit config' launch file

name: hand_controller
action_ns: gripper_action
type: GripperCommand
default: true
parallel: true
joints:
panda_finger_joint1
panda_finger_joint2

Then, I wrote 'my robot controller.yaml file' as follows.

hand_controller:
type: position_controllers/GripperActionController
joints:
- panda_finger_joint1
- panda_finger_joint2
...(omit below)

I expected a connection between moveit and gazbo's gripper controller, just like panda_arm_controller (FollowJointTrajectory)

However, moveit and gazbo's controller are not connected.

I want to use gripper in Gazebo or Real robot by moveit.

I would appreciate it if you could send me some advice

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which git commit or tag? Kinetic version

v4hn, 2020-06-23:
Well, what topics *did* you get? check leaf topics, e.g., via `rqt_graph`.

```
 action_ns: gripper_action
```
From [the source](https://github.com/ros-controls/ros_controllers/blob/noetic-devel/gripper_action_controller/include/gripper_action_controller/gripper_action_controller_impl.h#L214) it looks like the controller advertises `gripper_cmd` instead.

seoulstudent, 2020-06-23:
Dear @v4hn 

Thank you for your prompt reply.  Your advice was correct
I'd like to ask one more question.

'GripperActionController' only use one joint. 
Even if Moveit sends two joints value (panda_finger_joint1&panda_finger_joint2) only one can be used.

Also, FollowJointTrajectory doesn't get along with moveit_msgs::Grasp.  

I would appreciate it if you could send me some advice. 
(If I want to use moveit_msgs::Grasp, should I make ros_controller separately?)

seoulstudent, 2020-07-06:
If you are solving the same problem as me, please refer to the following link.
https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_control/config/j2s6s300_control.yaml

closed 2020-07-06 by seoulstudent