moveit/moveit_tutorials #505: Doesn't show octomap in Perception pipeline tutorial

huiwenzhang, 2020-06-30:
### Description

I followed instructions in http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/perception_pipeline/perception_pipeline_tutorial.html. The launch file should spawn the rviz and  a franka arm and the octomap show. However, only arm is correctly displayed, no octomap.

### Your environment
* ROS Distro: [Melodic]
* OS Version: Ubuntu 18.04
* Source build with master branch

### Steps to reproduce
To reproduce, just follow instructions in http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/perception_pipeline/perception_pipeline_tutorial.html.

### Expected behaviour
The franka arm shows in rviz with an Octomap represents the obstacles in the environment.

### Backtrace or Console output

When I run `roslaunch moveit_tutorials obstacle_avoidance_demo.launch`, the console outputs
```
started roslaunch server http://alvin-X280:34129/
SUMMARY
========
PARAMETERS
 * /joint_state_publisher/use_gui: False
 * /move_group/allow_trajectory_execution: True
 * /move_group/controller_list: [{'joints': ['pan...
 * /move_group/hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/initial: [{'pose': 'ready'...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_fake_contr...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_frame: camera_rgb_optica...
 * /move_group/octomap_resolution: 0.25
 * /move_group/panda_arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/panda_arm_hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/planner_configs/BFMTkConfigDefault/balanced: 0
 * /move_group/planner_configs/BFMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/BFMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/BFMTkConfigDefault/heuristics: 1
 * /move_group/planner_configs/BFMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/BFMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/BFMTkConfigDefault/optimality: 1
 * /move_group/planner_configs/BFMTkConfigDefault/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMTkConfigDefault/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/BiESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiESTkConfigDefault/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRTkConfigDefault/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/init_temperature: 100
 * /move_group/planner_configs/BiTRRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/type: geometric::BiTRRT
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/FMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/FMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/FMTkConfigDefault/heuristics: 0
 * /move_group/planner_configs/FMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/FMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/FMTkConfigDefault/radius_multiplier: 1.1
 * /move_group/planner_configs/FMTkConfigDefault/type: geometric::FMT
[... 210 lines cut ...]
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1593526248.305266119]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1593526248.305289798]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1593526248.524273836]: Stereo is NOT SUPPORTED
[ INFO] [1593526248.524355554]: OpenGl version: 3 (GLSL 1.3).
[ INFO] [1593526248.693679293]: Loading robot model 'panda'...
[ INFO] [1593526249.072796106]: Loading robot model 'panda'...
[ INFO] [1593526249.156872495]: Starting planning scene monitor
[ INFO] [1593526249.159388537]: Listening to '/move_group/monitored_planning_scene'
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
```
I checked out the outputs, seems nothing going wrong here, but the rviz doesn't show octomap. I echoed the `/move_group/monitored_planning_scene` topic, nothing shows.

v4hn, 2020-06-30:
Did you run into the mesa bug [we only documented recently in master](https://ros-planning.github.io/moveit_tutorials/doc/perception_pipeline/perception_pipeline_tutorial.html#running-the-interface)?

huiwenzhang, 2020-07-01:
> Did you run into the mesa bug [we only documented recently in master](https://ros-planning.github.io/moveit_tutorials/doc/perception_pipeline/perception_pipeline_tutorial.html#running-the-interface)?

I set the environment variable as suggested by the link, but it doesn't work. I don't think this is caused by the rendering problem, since the arm is visualized correctly. The strange point is the topic `/move_group/monitored_planning_scene` seems to be empty, while  the pointcloud msg is published normally.

lianghongzhuo, 2021-04-15:
I have the save issues in moveit melodic version.  Only see Franka robot, no octomap.
Any updates on this issues?

lianghongzhuo, 2021-04-15:
`rostopic echo /camera/depth_registered/points` can output point msgs
but can not see the pointcloud using rviz

lianghongzhuo, 2021-04-15:
I have fixed the issue, The launch file is wrong. I will open a pull request to fix the bug.

rhaschke, 2021-04-15:
Fixed via backporting #491.

closed 2021-04-15 by rhaschke

lianghongzhuo, 2021-04-16:
Thanks for the fix.