moveit/moveit_tutorials #506: run panda_moveit_config demo with self collision on gripper

Wuqiqi123, 2020-07-05:
### Description

Overview of your issue here.

### Your environment
* ROS Distro:  ```* /rosdistro: melodic * /rosversion: 1.14.3```
* OS Version: e.g. Linux 5.6.16-1-MANJARO
* Source or Binary build? Source git branch of melodic-devel at 5c3090b00dd0b9eb9cf73ab2154c740eaf4a1b2d.

### self collision is wrong.
when I run demo ```roslaunch panda_moveit_config demo.launch rviz_tutorial:=true``` according to [https://ros-planning.github.io/moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html](url).

[image]

Is collision configuration is set wrong?

rhaschke, 2020-07-06:
Obviously, you added capsule-like collision objects (the green ones) and rviz shows a collision in the last links (red highlights).
So, what exactly are you complaining about? Please, more precisely describe your issue. Provide details how you added the collision objects. Where do you see unexpected behaviour?

hybrayhem, 2021-03-14:
I reproduced issue in my Melodic setup on Ubuntu 18.04. Capsule-like collisions comes from [Franka description package](https://github.com/frankaemika/franka_ros/tree/kinetic-devel/franka_description) that recommended to use [here ](http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html#getting-started)(moveit melodic tutorials). I think you can solve the problem with replacing capsule-like collisions with mesh of hand like [this commit](https://github.com/hybrayhem/franka_ros/commit/25c8ff580917552c77d1442fe52f5203459dfc4a).
Universal robots also add collisions [in this way](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_description/urdf/ur3.urdf.xacro#L82-L86). 

The difference is as follows:
- **Panda arm with mesh collisions**

```xml
<collision>
  <geometry>
    <mesh filename="package://franka_description/meshes/visual/hand.dae"/>
  </geometry>
</collision>
```

- **Panda arm with capsule-like collisions**
```xml
<collision>
  <origin xyz="0 0 0.04" rpy="0 ${pi/2} ${pi/2}"/>
  <geometry>
    <cylinder radius="${0.04+safety_distance}" length="0.1" />
  </geometry>
</collision>
<collision>
  <origin xyz="0 -0.05 0.04" rpy="0 0 0"/>
  <geometry>
    <sphere radius="${0.04+safety_distance}"  />
  </geometry>
</collision>
<collision>
  <origin xyz="0 0.05 0.04" rpy="0 0 0"/>
  <geometry>
    <sphere radius="${0.04+safety_distance}"  />
  </geometry>
</collision>
<collision>
  <origin xyz="0 0 0.1" rpy="0 ${pi/2} ${pi/2}"/>
  <geometry>
    <cylinder radius="${0.02+safety_distance}" length="0.1" />
  </geometry>
</collision>
<collision>
  <origin xyz="0 -0.05 0.1" rpy="0 0 0"/>
  <geometry>
    <sphere radius="${0.02+safety_distance}"  />
  </geometry>
</collision>
<collision>
  <origin xyz="0 0.05 0.1" rpy="0 0 0"/>
  <geometry>
    <sphere radius="${0.02+safety_distance}"  />
  </geometry>
</collision>
```