moveit/moveit_tutorials #517: Running MoveItCpp tutorial with a gazebo controller results in Trajectory messages that are not strictly increasing in time

rishabhgks, 2020-08-20:
### Description

I am trying to use the MoveItCpp interface to control a [Staubli TX-60](https://github.com/mwamp/staubli_experimental) arm. I use their JointTrajectoryController and set the "moveit_controller_manager" to MoveItSimpleControllerManager. The planning happens perfectly and a plan is generated, but when I run the execute() function I get the error:
 ```
Waiting to continue: Press 'next' in the RvizVisualToolsGui window to start the demo... continuing
[ WARN] [1597949866.466092505, 6.338000000]: It looks like the planning volume was not specified.
[ INFO] [1597949866.468209561, 6.339000000]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1597949866.470186891, 6.341000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1597949866.492377495, 6.359000000]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1597949866.492431530, 6.359000000]: Solution found in 0.023068 seconds
[ INFO] [1597949866.512285198, 6.372000000]: SimpleSetup: Path simplification took 0.019639 seconds and changed from 4 to 2 states
[ERROR] [1597949866.539251212, 6.395000000]: Trajectory message contains waypoints that are not strictly increasing in time.
[ WARN] [1597949866.542895709, 6.398000000]: Controller arm_controller failed with error INVALID_GOAL: Trajectory message contains waypoints that are not strictly increasing in time.
[ WARN] [1597949866.543342958, 6.398000000]: Controller handle arm_controller reports status FAILED
[ INFO] [1597949866.543531351, 6.398000000]: Completed trajectory execution with status FAILED ...
```
I tried to debug further and came at the point where planning_component returns a solution to the moveit_cpp_tutorial. I converted the plan_solution1->trajectory to moveit_msgs/RobotTrajectory msg and printed it out on a topic. This is an example of the type of messages being passed by the planner to the TrajectoryExecutionManager:
```
trajectory: 
  - 
    joint_trajectory: 
      header: 
        seq: 0
        stamp: 
          secs: 0
          nsecs:         0
        frame_id: "world"
      joint_names: [joint_1, joint_2, joint_3, joint_4, joint_5, joint_6]
      points: 
        - 
          positions: [1.504574242972012e-11, -3.284306160367123e-11, -9.605649609056854e-12, -2.1325163857000007e-12, 1.0311751452718454e-12, 8.455458555545192e-13]
          velocities: []
          accelerations: []
          effort: []
          time_from_start: 
            secs: 0
            nsecs:         0
        - 
          positions: [-7.880131335270788e-05, 0.15267103477495161, -0.1833828864002482, -0.0040051460742771645, 0.03048874774064969, 0.003952845536801455]
          velocities: []
          accelerations: []
          effort: []
          time_from_start: 
            secs: 0
            nsecs:         0
        - 
          positions: [-0.0001576026417511582, 0.3053420695827463, -0.36676577279089073, -0.008010292146421813, 0.06097749548026821, 0.007905691072757364]
          velocities: []
          accelerations: []
          effort: []
          time_from_start: 
            secs: 0
            nsecs:         0
        - 
          positions: [-0.0002364039701496085, 0.458013104390541, -0.5501486591815332, -0.012015438218566462, 0.09146624321988672, 0.011858536608713272]
          velocities: []
          accelerations: []
          effort: []
          time_from_start: 
            secs: 0
            nsecs:         0
        - 
          positions: [-0.0003152052985480588, 0.6106841391983356, -0.7335315455721758, -0.01602058429071111, 0.12195499095950524, 0.01581138214466918]
          velocities: []
          accelerations: []
          effort: []
          time_from_start: 
            secs: 0
            nsecs:         0
        - 
[... 271 lines cut ...]
    // Visualize the trajectory in Rviz
    visual_tools.publishTrajectoryLine(plan_solution1.trajectory, joint_model_group_ptr);
    visual_tools.trigger();

    /* Uncomment if you want to execute the plan */
    planning_components->execute(); // Execute the plan */
  }

  // Start the next plan
  visual_tools.deleteAllMarkers();
  visual_tools.prompt("Press 'next' in the RvizVisualToolsGui window to continue the demo");

  // END_TUTORIAL
  visual_tools.deleteAllMarkers();
  visual_tools.prompt("Press 'next' to end the demo");

  ROS_INFO_STREAM_NAMED(LOGNAME, "Shutting down.");
  ros::waitForShutdown();
}
```
Please let me know If I can provide any more details on the issue. I'm not familiar with dockers otherwise I would have uploaded one instead of pasting the code.

### Expected behaviour
I believe that I am missing some kind of parameter that is not calculating the time, velocities and accelerations for each waypoint inside the planned trajectory. I could be wrong.
Ultimately I want the arm to move with gazebo controllers and see the movement happen in gazebo as well so that I can plan continuous movements using moveitcpp.

rishabhgks, 2020-08-24:
@JafarAbdi @henningkayser @rhaschke 
Could any of you please guide me on where I might be going wrong here?

corot, 2021-06-20:
I'm experiencing the same issue when picking/placing in gazebo, but only with the current melodic.
Doesn't happen if I use compiled master
```
...
      - 
        positions: [1.2889687027773171, 0.055803899290690184, -0.6248854557842535, 0.7660033740710387]
        velocities: []
        accelerations: []
        effort: []
        time_from_start: 
          secs: 0
          nsecs:         0
      - 
        positions: [1.2434171380344599, 0.15359409387309908, -0.6561297106314428, 0.7343035428209158]
        velocities: []
        accelerations: []
        effort: []
        time_from_start: 
          secs: 0
          nsecs:         0
      - 
        positions: [1.1978655732916028, 0.25138428845550775, -0.687373965478632, 0.7026037115707929]
        velocities: []
        accelerations: []
        effort: []
        time_from_start: 
          secs: 0
          nsecs:         0
      - 
        positions: [1.1523140085487455, 0.34917448303791687, -0.7186182203258215, 0.6709038803206699]
        velocities: []
        accelerations: []
        effort: []
        time_from_start: 
          secs: 0
...
```
Funny thing is that this started happening after re-making my moveit configuration with an updated wizard!
(here the [update commit](https://github.com/corot/thorp/commit/bac6e29e695a5c8ef7090e3447fae74cc567793b))