moveit/moveit_tutorials #525: MoveIt Tutorials Noetic Release (<VERSION>)

tylerjw, 2020-09-14:
# MoveIt Tutorials Noetic Release (<VERSION>)

Does anyone know if there is anything extra that needs to happen to cut a release of the tutorials because it builds into a website so we get a new noetic section to the website?

### Backports
- [ ] .
- [ ] .

### Checklist
- [ ] Merge backports
- [ ] Run ROS buildfarm prerelease test
- [ ] Update changelogs
- [ ] Bloom
- [ ] Write discourse post

rhaschke, 2020-09-15:
I don't know how the release process is handled for the tutorial pages. However, we are still missing `franka_description` for a release, aren't we?

tylerjw, 2020-09-30:
It looks like it was released: https://index.ros.org/p/franka_description/#noetic

tylerjw, 2020-09-30:
This is blocked by https://github.com/ros-planning/panda_moveit_config/issues/72

DLu, 2020-10-07:
Also blocked by https://github.com/ros-industrial/universal_robot/issues/537 and https://github.com/ros-planning/moveit_grasps/issues/100

dqii, 2020-11-11:
Does the tutorial still work with Noetic, or would it be better to use Melodic for now?

science-code, 2022-07-08:
When following the ROS Noetic version of the [grasping tutorial](https://ros-planning.github.io/moveit_tutorials/doc/moveit_grasps/moveit_grasps_tutorial.html), cloning the `moveit_grasps` repository fails:
```
git clone -b $ROS_DISTRO-devel https://github.com/ros-planning/moveit_grasps.git
Cloning into 'moveit_grasps'...
fatal: Remote branch noetic-devel not found in upstream origin
```

rhaschke, 2022-07-08:
Obviously, there is no noetic-devel branch. Try the master branch. If that works, please file a PR to fix the tutorial. Thanks.

ImSubin98, 2023-11-21:
hi all, is anybody solve this problem?
(in ubuntu 20.04, noetic ros1)

rickstaa, 2023-11-21:
> hi all, is anybody solve this problem? (in ubuntu 20.04, noetic ros1)

Hey @ImSubin98, I had the intention to shift the tutorials to Noetic a while back, but life threw a curveball (check out https://github.com/ros-planning/panda_moveit_config/issues/97). If anyone's up for it, feel free to tackle the issue. On another note, @rhaschke and I already sorted out the initial item on @tylerjw's list a while back.