moveit/moveit_tutorials #527: MoveIt Setup Assistant crashes while trying to set Virtual joint

AnlageM, 2020-09-17:
### Description

 MoveIt Setup Assistant crashes while trying to set Virtual joint 

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source build

### Steps to reproduce
Hi, I installed Moveit and its dependencies on ROS Noetic, and I am trying to follow this tutorial:

https://ros-planning.github.io/moveit...

I have both tried with panda_arm_hand.urdf.xacro and other .urdf or .xacro or .dae: no problem in generating collision matrix, but when I try to add the virtual joint the Setup Assistant window is closed with this error:

[ERROR] [1600333582.658529467]: Could not find the 'robot' element in the xml file
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to update the SRDF Model
================================================================================REQUIRED process [moveit_setup_assistant-2] has died!
process has died [pid 3186, exit code -6, cmd /home/aha/ws_IKtrial/devel_isolated/moveit_setup_assistant/lib/moveit_setup_assistant/moveit_setup_assistant __name:=moveit_setup_assistant __log:=/home/aha/.ros/log/afeb18c8-f8c1-11ea-bcd2-75be2c1310bf/moveit_setup_assistant-2.log].
log file: /home/aha/.ros/log/afeb18c8-f8c1-11ea-bcd2-75be2c1310bf/moveit_setup_assistant-2*.log
Initiating shutdown!
================================================================================

Opening the file, I can definitely see the robot tag. So why does this error happen??

rhaschke, 2020-09-17:
Are you using srdfdom 0.6.2? There was a recent fix in srdfdom, which exactly fixes this problem.

closed 2020-09-17 by rhaschke

AnlageM, 2020-09-17:
Thanks for the help, indeed using srdfdom 0.6.2 it works. Just a check, am I supposed to install srdfdom 0.6.2 from source, right? By installing the noetic binary, I could get only 0.6.0, I ask because I am new to ros and git, and I want to ensure I install packages properly.

rhaschke, 2020-09-18:
New binary releases are currently only available in [`ros-testing`](http://wiki.ros.org/action/show/TestingRepository?action=show&redirect=ShadowRepository). Only with the next sync, they will become available in the standard repo as well.