moveit/moveit_tutorials #531: Default Planning Groups Image (and _Kin. parameters file_ field)

tylerjw, 2020-09-30:
### Description

Following the tutorial on master for creating the moveit_panda_config I noticed that the image for the `Planning Groups` tab doesn't match the current version of that page.  There is a new parameter called _Kin. parameters file_.  There are no instructions for this field also.

Reference page: https://ros-planning.github.io/moveit_tutorials/doc/setup_assistant/setup_assistant_tutorial.html#step-4-add-planning-groups

New image to compare (locally there is some issue where the robot isn't showing in the right pane, I'm doing this on Noetic on the master of moveit):
[image]

Ctiger96, 2020-12-13:
I have the same puzzle, would you plz tell me what to do with it?

YANGHANJIANG, 2020-12-25:
I have the same puzzle. Any help would be appreciated.

mlautman, 2021-02-25:
Hey everyone, thank you for bringing this to our attention! MoveIt is a community project that is constantly improving over time so it isn't surprising to find new features coming out where documentation might lag. I searched through the ros-planning/moveit repo to find the additional feature here https://github.com/ros-planning/moveit/pull/1997. (Thanks @v4hn for putting that together!) If any of you @YANGHANJIANG @Ctiger96  or @tylerjw (Hi Tyler!) could update the tutorials to document the new features that would be awesome! It shouldn't take much effort. Just an updated png in the moveit_tutorials repo and some additional text in the rst. :)

Sophoverick, 2022-10-14:
If you followed the [Getting started tutorials](http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/getting_started/getting_started.html) that you were supposed to follow prior to this one then the kinematics file you are looking for should have been dowloaded. The directory of the kinematics file would be ~/ws_moveit/src/panda_moveit_config/config/kinematics.yaml, you would find [this file](https://github.com/ros-planning/panda_moveit_config/blob/noetic-devel/config/kinematics.yaml).