moveit/moveit_tutorials #533: [setup assistant] Robot disapears when you add the virtual joint

tylerjw, 2020-09-30:
### Description

I don't know if this is intentional but when you add the fixed virtual joint to world the robot model disapears from the right side.

This was done with franka_description installed from released debian on Noetic and MoveIt built from master at commit 6eeee4b7a.

Here is a couple images to show you what I mean:
[image]
[image]

Here is the console output:
```bash
$ roslaunch moveit_setup_assistant setup_assistant.launch 
... logging to /home/tyler/.ros/log/ede7cbc8-034b-11eb-b3f0-69d0e0971aef/roslaunch-eboladrome-91643.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://eboladrome:43719/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.8

NODES
  /
    moveit_setup_assistant (moveit_setup_assistant/moveit_setup_assistant)

auto-starting new master
process[master]: started with pid [91651]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ede7cbc8-034b-11eb-b3f0-69d0e0971aef
process[rosout-1]: started with pid [91661]
started core service [/rosout]
process[moveit_setup_assistant-2]: started with pid [91664]
ros.moveit_setup_assistant: checking in /opt/ros/noetic/share/franka_description/robots/panda_arm_hand.urdf.xacro
ros.moveit_setup_assistant: checking in /opt/ros/noetic/share/franka_description/robots
ros.moveit_setup_assistant: checking in /opt/ros/noetic/share/franka_description
ros.moveit_setup_assistant: Found package.xml in /opt/ros/noetic/share/franka_description
ros.moveit_setup_assistant: Package name for file "/opt/ros/noetic/share/franka_description/robots/panda_arm_hand.urdf.xacro" is "franka_description"
ros.moveit_setup_assistant: URDF Package Name: franka_description
ros.moveit_setup_assistant: URDF Package Path: robots/panda_arm_hand.urdf.xacro
ros.moveit_setup_assistant: Loaded panda robot model.
ros.moveit_setup_assistant: Setting Param Server with Robot Description
ros.moveit_setup_assistant: Robot semantic model successfully loaded.
ros.moveit_setup_assistant: Setting Param Server with Robot Semantic Description
ros.moveit_core.robot_model: Loading robot model 'panda'...
ros.moveit_core.robot_model: No root/virtual joint specified in SRDF. Assuming fixed joint
ros.moveit_core.robot_model.empty_collision_geometry: Link panda_leftfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
ros.moveit_core.robot_model.empty_collision_geometry: Link panda_rightfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
ros.rviz: Stereo is NOT SUPPORTED
ros.rviz: OpenGl version: 4.6 (GLSL 4.6).
ros.moveit_core.robot_model: Loading robot model 'panda'...
ros.moveit_core.robot_model: No root/virtual joint specified in SRDF. Assuming fixed joint
ros.moveit_core.robot_model.empty_collision_geometry: Link panda_leftfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
ros.moveit_core.robot_model.empty_collision_geometry: Link panda_rightfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
ros.moveit_setup_assistant: Loading Setup Assistant Complete
ros.moveit_setup_assistant: -------------------------------------------------------------------------------
ros.moveit_setup_assistant: Statistics:
ros.moveit_setup_assistant:     12 : Total Links
ros.moveit_setup_assistant:     66 : Total possible collisions
ros.moveit_setup_assistant:      0 : Always in collision
ros.moveit_setup_assistant:     11 : Never in collision
ros.moveit_setup_assistant:      4 : Default in collision
ros.moveit_setup_assistant:      9 : Adjacent links disabled
ros.moveit_setup_assistant:     42 : Sometimes in collision
ros.moveit_setup_assistant:     24 : TOTAL DISABLED
ros.moveit_setup_assistant: Thread complete 45
ros.moveit_setup_assistant: Updating kinematic model
ros.moveit_core.robot_model: Loading robot model 'panda'...
ros.moveit_core.robot_model.empty_collision_geometry: Link panda_leftfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
ros.moveit_core.robot_model.empty_collision_geometry: Link panda_rightfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
ros.moveit_core.robot_model: Loading robot model 'panda'...
ros.moveit_core.robot_model: No root/virtual joint specified in SRDF. Assuming fixed joint
ros.moveit_core.robot_model.empty_collision_geometry: Link panda_leftfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
ros.moveit_core.robot_model.empty_collision_geometry: Link panda_rightfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
```

tylerjw, 2020-09-30:
I also tested this with the version of franka_description in their repo which is of the same version as the one released into Noetic debians (0.7.0).

rhaschke, 2020-10-01:
This is an issue in MSA itself. I filed a fix: https://github.com/ros-planning/moveit/pull/2335

closed 2020-10-01 by rhaschke