moveit/moveit_tutorials #548: Improve the tutorials and documentation

kappa95, 2020-10-21:
## Improve the documentation of the tutorials for Python
There are big differences in the documentation between Python and C++ and so could be better Improve and explain more in detail the features present in programming with Python. Particularly in basics operations like pick and place

### Add the documentation about multiple robot
It is missing a part related how to move multiple robots, there are situations in which we have robots with multiple arms or we need multiple manipulators. Could be interesting and useful if you add these kind of topics

## Improve the documentation of the OMPL planners
Is it possible to implement and explain more in detail the topic of the planners:
How they works in Moveit, their behaviours and how to use them to predict a little the solutions of the trajectory planning.

davetcoleman, 2020-11-12:
Please contribute these @kappa95 ! :-)

kappa95, 2020-11-12:
> Please contribute these @kappa95 ! :-)

I'm still discovering how to do the features that I asked 😂🖖

predator4hack, 2020-12-24:
@davetcoleman I have written a python script for pick and place but after grabbing the object, I'm getting this error: Solution found but the environment changed during execution and the path was aborted.
Can you suggest any way to solve this? I referred to the issue: https://github.com/ros-planning/moveit/issues/417 but didn't get any solution.

felixvd, 2021-02-10:
@kappa95 Did you figure out some of the things on this list? Will you be able to write something up? People who recently suffered from tutorial shortcomings are usually the best people to do it.

@predator4hack Try posting a detailed question on ROS Answers for your problem.

kappa95, 2021-02-10:
> @kappa95 Did you figure out some of the things on this list? Will you be able to write something up? People who recently suffered from tutorial shortcomings are usually the best people to do it.
> 
> @predator4hack Try posting a detailed question on ROS Answers for your problem.

@felixvd Hi, well actually reading a lot of posts on the web I discovered something of the list:
* Regarding moving robot with two arms (but surely works even with more arms): if i configure the group of both arms from the setup assistant with joints instead of chain. I can use the group of two arms in order to set two different target in this way: `group_both.set_pose_target(point1, "End effector arm 1")` and `group_both.set_pose_target(point2, "End effector arm 2")` and later using plan and execute they move together.
* Instead regarding multiple robots (more robot in the same URDF), I haven't found a way yet. But theoretically if you use them as a single robot and configure the two robots as joints as for the two arms, it should work.
* Regarding the OMPL and planners in general, I read a little by the official documentation on the [OMPL](http://ompl.kavrakilab.org/) but actually i don't know how to choose a planner basing on the application. Generally I set only RRTConnect which seems the fastest.

Actually I'm still a newbie about ROS, I'm trying to write my master thesis and one of my advisor talk me about ROS and i'm using it on a Yumi. And for that i'm searching answers and good tutorials for learning.

mlautman, 2021-02-25:
@kappa95

closed 2021-02-25 by mlautman