moveit/moveit_tutorials #564: Cannot visualize MotionPlanning in RViz

SiHaoShen, 2020-12-09:
### Description

Hi, 

I am trying to use the tutorial to understand Moveit, however I am stucked in the tutorial "MoveIt Quickstart in RViz". Since I can't open "MotionPlanning" from the moveit_ros_visualization folder, I waited more than half an hour, RViz won't responding. Restarting my PC or rebuilding the tutorial from source doesn't work either.
[image]

Here is the link where I stuck now in the tutorial: http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source

### Steps to reproduce
Follow the getting started steps to install moveit melodic and setup moveit workspace. Then follow steps from MoveIt Quickstart in RViz.

### Expected behaviour
After choosing "MotionPlanning" as the DislayType, the Panda robot should be visible in RViz.

### Backtrace or Console output

```
ROS_MASTER_URI=http://localhost:11311

process[virtual_joint_broadcaster_1-1]: started with pid [26008]
process[joint_state_publisher-2]: started with pid [26009]
process[joint_state_desired_publisher-3]: started with pid [26010]
process[robot_state_publisher-4]: started with pid [26015]
process[move_group-5]: started with pid [26017]
process[rviz_isaactitan_desktop_25984_6312870121317032954-6]: started with pid [26022]
[ INFO] [1607521971.270334392]: Loading robot model 'panda'...
[ INFO] [1607521971.321716484]: rviz version 1.13.15
[ INFO] [1607521971.321753124]: compiled against Qt version 5.9.5
[ INFO] [1607521971.321763072]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1607521971.326556099]: Publishing maintained planning scene on 'monitored_planning_scene'
[ INFO] [1607521971.330279333]: MoveGroup debug mode is ON
Starting planning scene monitors...
[ INFO] [1607521971.330310341]: Starting planning scene monitor
[ INFO] [1607521971.332904894]: Listening to '/planning_scene'
[ INFO] [1607521971.332941343]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[ INFO] [1607521971.334021063]: Forcing OpenGl version 0.
[ INFO] [1607521971.335166928]: Listening to '/collision_object'
[ INFO] [1607521971.336010832]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] [1607521971.344540518]: Listening to '/camera/depth_registered/points' using message filter with target frame 'world '
[ INFO] [1607521971.347429266]: Listening to '/attached_collision_object' for attached collision objects
Planning scene monitors started.
[ INFO] [1607521971.364384832]: Initializing OMPL interface using ROS parameters
[ INFO] [1607521971.389516476]: Using planning interface 'OMPL'
[ERROR] [1607521971.392233560]: Exception while loading planning adapter plugin 'default_planner_request_adapters/ResolveConstraintFrames': According to the loaded plugin descriptions the class default_planner_request_adapters/ResolveConstraintFrames with base class type planning_request_adapter::PlanningRequestAdapter does not exist. Declared types are  default_planner_request_adapters/AddIterativeSplineParameterization default_planner_request_adapters/AddTimeOptimalParameterization default_planner_request_adapters/AddTimeParameterization default_planner_request_adapters/Empty default_planner_request_adapters/FixStartStateBounds default_planner_request_adapters/FixStartStateCollision default_planner_request_adapters/FixStartStatePathConstraints default_planner_request_adapters/FixWorkspaceBounds
[ INFO] [1607521971.392589803]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1607521971.392842691]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1607521971.393091160]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1607521971.393351362]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1607521971.393665495]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1607521971.393937569]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1607521971.393981262]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1607521971.393997853]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1607521971.394014314]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1607521971.394030475]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1607521971.394045804]: Using planning request adapter 'Fix Start State Path Constraints'
[ INFO] [1607521971.401910143]: Loading robot model 'panda'...
[ INFO] [1607521971.427159690]: Set joints of group 'panda_arm' to pose 'ready'.
[ INFO] [1607521971.427188354]: Set joints of group 'hand' to pose 'open'.
[ INFO] [1607521971.427401687]: Fake controller 'fake_panda_arm_controller' with joints [ panda_joint1 panda_joint2 panda_joint3 panda_joint4 panda_joint5 panda_joint6 panda_joint7 ]
[ INFO] [1607521971.427577850]: Fake controller 'fake_hand_controller' with joints [ panda_finger_joint1 panda_finger_joint2 ]
[ INFO] [1607521971.427817262]: Returned 2 controllers in list
[ INFO] [1607521971.433116084]: Trajectory execution is managing controllers
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
[ INFO] [1607521971.465399991]: 

********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1607521971.465462580]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1607521971.465488559]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1607521971.874229238]: Stereo is NOT SUPPORTED
[ INFO] [1607521971.874294572]: OpenGl version: 4,6 (GLSL 4,6).

```

JafarAbdi, 2020-12-09:
There's an exception being thrown while loading planning adapters, please make sure to source your workspace `source PATH_TO_WS/devel/setup.bash`

```
[ERROR] [1607521971.392233560]: Exception while loading planning adapter plugin 'default_planner_request_adapters/ResolveConstraintFrames': According to the loaded plugin descriptions the class default_planner_request_adapters/ResolveConstraintFrames with base class type planning_request_adapter::PlanningRequestAdapter does not exist. Declared types are  default_planner_request_adapters/AddIterativeSplineParameterization default_planner_request_adapters/AddTimeOptimalParameterization default_planner_request_adapters/AddTimeParameterization default_planner_request_adapters/Empty default_planner_request_adapters/FixStartStateBounds default_planner_request_adapters/FixStartStateCollision default_planner_request_adapters/FixStartStatePathConstraints default_planner_request_adapters/FixWorkspaceBounds
```

SiHaoShen, 2020-12-09:
Thank you for the reply! However, I really source the setup bash file to my workspace, as the tutorial suggested. And the problem

This is what I have done until now:

```
rosdep update
sudo apt-get update
sudo apt-get dist-upgrade

sudo apt-get install ros-melodic-catkin python-catkin-tools

sudo apt install ros-melodic-moveit
mkdir -p ~/ws_moveit/src

cd ~/ws_moveit/src
git clone https://github.com/ros-planning/moveit_tutorials.git -b melodic-devel
git clone https://github.com/ros-planning/panda_moveit_config.git -b melodic-devel

cd ~/ws_moveit/src
rosdep install -y --from-paths . --ignore-src --rosdistro melodic

cd ~/ws_moveit
catkin config --extend /opt/ros/${ROS_DISTRO} --cmake-args -DCMAKE_BUILD_TYPE=Release
catkin build

source ~/ws_moveit/devel/setup.bash

echo 'source ~/ws_moveit/devel/setup.bash' >> ~/.bashrc

roslaunch panda_moveit_config demo.launch rviz_tutorial:=true
```

When I build the from here
`git clone https://github.com/ros-planning/moveit_tutorials.git -b master`

The error message is gone, however the same problem still exist:

```
process[virtual_joint_broadcaster_1-1]: started with pid [24321]
process[joint_state_publisher-2]: started with pid [24322]
process[joint_state_desired_publisher-3]: started with pid [24323]
process[robot_state_publisher-4]: started with pid [24329]
process[move_group-5]: started with pid [24334]
process[rviz_isaactitan_desktop_24297_174766000185833624-6]: started with pid [24336]
[ INFO] [1607525141.259571891]: Loading robot model 'panda'...
[ WARN] [1607525141.260238236]: Link panda_leftfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1607525141.260268723]: Link panda_rightfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ INFO] [1607525141.308560871]: rviz version 1.13.15
[ INFO] [1607525141.308613901]: compiled against Qt version 5.9.5
[ INFO] [1607525141.308628138]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1607525141.312493164]: Publishing maintained planning scene on 'monitored_planning_scene'
[ INFO] [1607525141.314914012]: MoveGroup debug mode is ON
Starting planning scene monitors...
[ INFO] [1607525141.314941994]: Starting planning scene monitor
[ INFO] [1607525141.317299072]: Listening to '/planning_scene'
[ INFO] [1607525141.317330651]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[ INFO] [1607525141.320049350]: Listening to '/collision_object'
[ INFO] [1607525141.322807452]: Forcing OpenGl version 0.
[ INFO] [1607525141.322931245]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] [1607525141.330090754]: Listening to '/camera/depth_registered/points' using message filter with target frame 'world '
[ INFO] [1607525141.333119606]: Listening to '/attached_collision_object' for attached collision objects
Planning scene monitors started.
[ INFO] [1607525141.375113406]: Using planning interface 'OMPL'
[ INFO] [1607525141.377129211]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1607525141.377317646]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1607525141.377482146]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1607525141.377668266]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1607525141.377842815]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1607525141.378013556]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1607525141.378045657]: Using planning request adapter 'Add Time Parameterization'
[... 25 lines cut ...]
[ INFO] [1607525141.447650719]: 

********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1607525141.447698058]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1607525141.447714238]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1607525141.848494229]: Stereo is NOT SUPPORTED
[ INFO] [1607525141.848561957]: OpenGl version: 4,6 (GLSL 4,6).
```

rhaschke, 2020-12-09:
If you are runing on Melodic with default debian MoveIt packages installed (that's what I got from your comments), then the melodic branches are the correct ones. The error message about unavailable `ResolveConstraintFrames` plugin can be ignored.

From your initial description, I understand that the rviz display dialog doesn't close after selecting the MotionPlanning display, right? So, I guess the culprit is on the rviz side. To further debug this, please run rviz in a separate terminal. To this end do the following:

roslaunch panda_moveit_config planning_context.launch load_robot_description:=true
rosrun rviz rviz

Then try to load the following displays: RobotModel, PlanningScene, MotionPlanning - in this order. And report any problems loading them.

felixvd, 2021-02-10:
Ping @SiHaoShen

mlautman, 2021-02-25:
Closing because this is more than 2 weeks old. @SiHaoShen: If you still experience this issue, feel free to re-open and we can continue to discuss

closed 2021-02-25 by mlautman