moveit/moveit_tutorials #572: Joint Limits not showing in the joint_limits.yaml

ly19264, 2021-01-07:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04

### Steps to reproduce
Creating a moveit package from moveit setup assistant

### Expected behaviour
Joint Limits should be taken from the URDF and be displayed in the joint_limits.yaml

### Example current output:
joint_limits:
  joint0:
    has_velocity_limits: true
    max_velocity: 100
    has_acceleration_limits: false
    max_acceleration: 0

Missing max_position and min_position

mlautman, 2021-02-25:
That would be a really cool feature but unfortunately the moveit setup assistant does not provide that functionality. Joint limits in moveit are a little more complicated than you might think it should be. It's one of the rough edges in using the software. I would recommend reading through the code in moveit core for setting hard and soft limit to understand how the joint limits are set to ensure you are doing things correctly. Furthermore, if you end up coming up with a better solution for setting joint limits from urdf/yaml please don't hesitate to open a PR 

Thank you  :)

closed 2021-02-25 by mlautman

v4hn, 2021-02-26:
Entries in the joint_limits.yaml are essentially overrides for limits defined in the URDF. Because, among other nodes, the actual hardware controllers will only see the limits from the URDF and it can be misleading to set them in the yaml.
So feel free to submit a patch to write the position limits extracted from the URDF to the yaml as well (should be straight-forward), but they should be disabled by default.