moveit/moveit_tutorials #574: possible problem with orientation in all_close() check

mqnc, 2021-01-13:
https://github.com/ros-planning/moveit_tutorials/blob/fff33d2b73af1d07ba329bbb0fceb9cb1c48edc0/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py#L67

this will fail on opposite quaternions which represent the same orientation

or is it guaranteed that the w component is always positive?

YijiongLin, 2021-01-17:
In a unit quaternion, the w should always be a positive one.  See [here](https://en.wikipedia.org/wiki/Quaternions_and_spatial_rotation).

[image]

And the w here actually is the real part of the quaternion. You can read more in the method: `geometry_msgs.msg.Quaternion()`

rhaschke, 2021-01-18:
The only limitation for unit quaternions is to be normalized to one. Particularly, it is possible to represent the very same rotation by q and -q. So yes, @mqnc's objection is correct, that `all_close` will fail for quaternions q and -q, although they represent the same orientation. @mqnc, did you observe any issues due to this shortcoming?

mqnc, 2021-01-19:
@rhaschke yes, I observed the issue that I cannot sleep well anymore since I've seen this. This is a tutorial which is supposed to show people how things are done properly. They will use this code as a template and it will run in industrial and medical applications. I can provide a pull request to fix this.

mqnc, 2021-01-19:
https://github.com/ros-planning/moveit_tutorials/pull/577

mlautman, 2021-02-25:
closing now that #577 is merged

closed 2021-02-25 by mlautman