moveit/moveit_tutorials #575: Why do the tutorial source code need the "all_close" function

YijiongLin, 2021-01-17:
https://github.com/ros-planning/moveit_tutorials/blob/fff33d2b73af1d07ba329bbb0fceb9cb1c48edc0/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py#L56

Could anyone give me some ideas about this function? What's the purpose of this function?

Thanks in advance!

All the best,

felixvd, 2021-02-10:
It's a convenience function to compare the values in two lists. You can read `all_close` as `all_values_in_these_lists_are_close`. You can see it used [here](https://github.com/ros-planning/moveit_tutorials/blob/fff33d2b73af1d07ba329bbb0fceb9cb1c48edc0/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py#L220) to compare if the robot is close to the target after executing the motion.

closed 2021-02-10 by felixvd

YijiongLin, 2021-02-10:
Hi @felixvd ,

Thanks for the reply! I've thought about checking the pose but not sure since it is in simulation and such a step seems not that necessary. But yes it is always good to check with the target pose condition. 

All the best,