moveit/moveit_tutorials #578: OpenRAVE installation issue

CaesarLinxw, 2021-01-20:
### Description
I followed the OpenRAVE installation process in the MoveIt tutorial (http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/ikfast/ikfast_tutorial.html). When I went through the source install steps of OpenRAVE and hit $ make -j$(nproc). Errors raised. The make log is attached.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source Install
* git clone --branch latest_stable https://github.com/rdiankov/openrave.git

### Steps to reproduce
Just follow the steps in the tutorial.

### Expected behaviour
I don't know what to expect.

### Backtrace or Console output
https://gist.github.com/CaesarLinxw/875bd3a96c3c825cab717ceded32a4a9

Thank you for your help!

Appreciatively,
Caesar

rhaschke, 2021-01-20:
I suggest to follow the more modern tutorial of the master branch for IKfast: https://ros-planning.github.io/moveit_tutorials/doc/ikfast/ikfast_tutorial.html
This one uses a pre-installed docker image to run OpenRAVE, which is tested and works for sure.

CaesarLinxw, 2021-01-20:
@rhaschke Thank you for your reply! Does it matter the branch is for Noetic but my ROS version is Melodic?

rhaschke, 2021-01-20:
The docker-based IKfast code generator will work on any system.

CaesarLinxw, 2021-01-20:
@rhaschke Thanks! I will give it a try!

felixvd, 2021-02-10:
Is this solved @CaesarLinxw ?

CaesarLinxw, 2021-02-10:
@felixvd 

I tried steps in this link: https://github.com/crigroup/openrave-installation to install the OpenRAVE before trying the docker-based generator and it went well.

felixvd, 2021-02-10:
@rhaschke There's nothing keeping the new installation instructions from being backported to Melodic, right? I'll open a first-timer issue about it if not.

rhaschke, 2021-02-10:
Yes, please go ahead. Technically, there is no reason to apply the same approach in Melodic as well. However, I think @davetcoleman wanted to keep the old approach around.

felixvd, 2021-02-10:
It's still in the [Kinetic tutorials](http://docs.ros.org/en/kinetic/api/moveit_tutorials/html/doc/ikfast/ikfast_tutorial.html) and the history, so it won't be lost. I made another issue so I'll close this one.

closed 2021-02-10 by felixvd

davetcoleman, 2021-03-10:
I don't have an opinion anymore about "keeping the old approach around", feel free to replace it