moveit/moveit_tutorials #581: OMPL planning example available?

ndehio, 2021-01-25:
Where can I find a c++/python example that demonstrates how to use the OMPL planner?

For my specific use case with the panda robot, the planner should output a trajectory with a goal state that minimizes a custom, user-specified cost (or, in other words, that is feasible with respect to a custom, user-specified constraint). The goal state to be optimized will have a non-zero velocity.

I read the MoveIt tutorials, but it is not clear to me how to approach this problem exactly.
Thanks for your suggestions!

tylerjw, 2021-01-25:
There is this tutorial PR: https://github.com/ros-planning/moveit_tutorials/pull/518

That depends on this PR on moveit: https://github.com/ros-planning/moveit/pull/2273 

If that is what you are looking for you'll need to build moveit from source to follow that tutorial.  It is not currently merged in because it is not passing CI.  Currently, the CI failure may or may not be a result of this PR.  We have not had time to yet figure it out unfortunatly.

ndehio, 2021-01-25:
Thanks for your quick answer.
This interesting PR-example imposes a path constraint:
```
pcm = moveit_msgs.msg.PositionConstraint()
...
path_constraints = moveit_msgs.msg.Constraints()
...
path_constraints.position_constraints.append(pcm)
...
move_group.set_path_constraints(path_constraints)
```
In my case, I want to impose a constraint on the goal state only. Is that possible?

ndehio, 2021-01-27:
I will try to rephrase my question:

How can I plan a trajectory that ensures the feasibility of a user-defined constraint for at least one of the discrete steps?

JafarAbdi, 2021-01-28:
I'm not sure about the Python interface, but for the C++ one in the MotionPlanRequest you can set the [goal_constraints](https://github.com/ros-planning/moveit_msgs/blob/master/msg/MotionPlanRequest.msg)

ndehio, 2021-02-08:
Thanks for your comments!

closed 2021-02-08 by ndehio