moveit/moveit_tutorials #585: "Connect MoveIt to real robot" is hard to discover

felixvd, 2021-02-10:
More and more users just want to make their robot move. The tutorial does not lead them to [the right page](http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/controller_configuration/controller_configuration_tutorial.html), and that page seems like it could be less technical and have more guidance for complete newcomers.

Changing the name of that page and adding a reference to the [Quickstart page](https://ros-planning.github.io/moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html) (like "If you want to use a real robot, look here") would go a long way.

I'm assigning the wmd-candidate label, but this would be a good first timer issue as well.

Mat198, 2022-02-16:
Totally agree. It's even harder if you are using custom hardware in the robot.

cnaes93, 2022-09-14:
I found how to move the real panda robot in noetic version with moveit so I write it here maybe it will be useful for beginners like me.
First run the franka-control like:
`roslaunch panda_moveit_config franka_control.launch robot_ip:=172.16.0.2`
There can man move the robot with rviz or you can write a python code and move the robot with it. For example launch in other tab the tutorial python code :
`rosrun moveit_tutorials move_group_python_interface_tutorial.py`

vocdex, 2025-06-23:
Just to add to @cnaes93 comment, you need to press the e-stop button to have blue light activated at the base while franka_control is running. Otherwise, the robot doesn't move in real world