moveit/moveit_tutorials #589: nothing happend when "param set robot_link_command_frame"

omeranar1, 2021-03-05:
->rosparam set /servo_server/optional_parameter_namespace/robot_link_command_frame my_link
->rosparam get /servo_server/optional_parameter_namespace/robot_link_command_frame
<-my_link

does not change the movements.
--
but when i change the "robot_link_command_frame" line in "ur_simulated_config.yaml" file then call 
->roslaunch moveit_servo spacenav_teleop_tools.launch 

everything working good

AndyZe, 2021-03-05:
Hi @omeranar1,

That's just the way ros parameters (usually) work. They get loaded at initialization but (usually) changing them at runtime has no effect. If you need that functionality, you'd probably use `dynamic_reconfigure`.

The good news is, you can dynamically change the command frame of the SpaceNavigator at runtime and Servo will do the transformation for you.

Servo takes a geometry_msgs/TwistStamped message. Just change the `header.frame_id` field to whatever you please.

AndyZe, 2021-03-05:
I'm going to close the issue because I don't see anything to be done. Feel free to open a new one if you have a different issue.

AndyZe, 2021-03-05:
If you want, make a PR to the tutorial to improve documentation of changing the command frame at runtime. That would be great.

closed 2021-03-05 by AndyZe

v4hn, 2021-03-05:
> That's just the way ros parameters (usually) work.
> They get loaded at initialization but (usually) changing them at runtime has no effect.
> If you need that functionality, you'd probably use `dynamic_reconfigure`.

I disagree. Your "usually" is just a specific programming pattern. It's perfectly fine to look up parameters at runtime in your node, but it is associated with additional overhead. Because moveit_servo should run at high rates without delays, it is still a good idea to leave things as they are. Especially because you already pointed out an alternative :+1: